mysensors/NodeManager #346: LDR reverse not working

taltholtmann, 2018-04-12:
I have an LDR Sensor which is reporting the light level.
But  I cant get the reserve option to work. No matter if i set it to true/false. It´s always the same behaviour.

user2684, 2018-04-18:
Hi, tagging this as a bug and adding to the 1.8 queue for further testing. Do you have any sample code to share? Thanks!

user2684, 2018-04-29:
Double checking the code, both SensorLDR and SoilMoistureSensor calls setReverse() during onSetup(), this is why the behavior does not change unless called after node.setup(). I'm going to fix this by creating a variable to store this preference in the next version, thanks again for reporting this!

user2684, 2018-10-02:
Fixed by #415

PR #415 (2018-10-02): LDR Reverse Fix
* Moved all the calls to setOutputPercentage(), setReference(), setReverse(), etc. from setup() to the constructor in SensorAnalogInput, SensorLDR, SensorRain and SensorSoilMoisture so allo the user to customize them in the sketch (fixes #346)
closed 2018-10-02 by user2684