mysensors/NodeManager #347: Si7021 does not play nice with other sensors, e.g. battery, signal

ttytyper, 2018-04-12:
Trying to run NodeManager 1.7 on a SenseBender Micro. When I enable Si7021 without any other sensors, the node reports humidity and temperature just fine. But if I enable other sensors in addition to that, I get a bunch of invalid readings. Tested with various combinations of Si7021, battery, signal. In all cases the node worked fine without Si7021, but started reporting bogus when Si7021 entered the mix.

Here's a sample of the output on serial. I get new random child IDs and types for every reboot.

NodeManager v1.7
LIB V=2.2.0-rc.2 R=N E=X T=N A=A S=A B=-
BATTERY I=201 P=30 T=38
SI7021 I=1 P=6 T=0
SI7021 I=2 P=7 T=1
RADIO...OK
PRES I=0 T=0 D=
PRES I=0 T=0 D=
PRES I=11601 T=9019 D=D�␓�$␡�.A��D(��0␝��+���6␡V��␇
PRES I=0 T=0 D=
PRES I=24578 T=12803 D=D(��0␝��+���6␡V��␇
ES I=0 T=0 D=
ES I=4578 T=12803 D=D(��
PRES I=128 T=26242 D=
PRES I=21536 T=12861 D=2861 D=2861 D=2861 D=2861 D=2861 
PRES I=-6479 T=-3673 D=������������
PRES I=-1 T=-1 D=�␏
PRES I=-32640 T=-30208 D=
PRES I=-9030 T=11512 D=⤃�6
PRES I=3389 T=17674 D=
PRES I=0 T=366 D�
���e�O␒t��␗E@�␚ۛ|��␐3��Ůc�!y8␘]nkL��4UREF]��la�,��]6��␏s������6�A���#p�ƾ�)#O␑␟�␎-␇␞�]��␟���P��␑^␃�F�^
PRES I=-7695 T=-14026 D=␏/␝8
PRES I=0 T=0 D=
PRES I=-27135 T=263 D==␏/␝8
ES I=0 T=0 D=
ES I=-27135 T=263 D==␏/␝8
ES I=0

user2684, 2018-04-18:
Thanks for reporting the issue. Weird, problem seems to be related to the description. Do you have the sample code you are using to share? When you compile it, how much of your storage results free in %? Thanks

ttytyper, 2018-05-08:
Sorry for the slow response. Not sure how to best upload code examples, so decided to upload them as commits to a fork:

Not working (both battery and Si7021 enabled):
https://github.com/ttytyper/NodeManager/commit/3d66934aa04eeff6b10c9a4c9544e1a16dcaa3ca

Working (just the battery sensor enabled):
https://github.com/ttytyper/NodeManager/commit/59faa40f6b8074d68342b9f314af2f9a0dd98835

I've also included the PlatformIO build log, showing the library versions and the storage spent.

Is there anything else I can do to troubleshoot? Anything else you need from me?

In case the Si7021 library from SparkFun is the culprit, there's a different library that I've had success with (on non-NodeManager nodes): https://github.com/LowPowerLab/SI7021

user2684, 2018-05-09:
Thanks! Will give your code a try once I'll get to this issue

user2684, 2018-07-08:
I was reviewing your logs and just realized also in your first message it is not just the description as I stated above which is weird but all the numbers I see above are not supposed to be there, same for the PRES repeated so many times. I'd say there must be some sort of incompatibility between the Si7021  and the library I'm currently using. I'll switch to the one used natively on the Micro which is https://github.com/mysensors/MySensorsArduinoExamples/tree/master/libraries/SI7021. Once something will be ready to test I'll share with you here. Thanks again!

ttytyper, 2018-10-03:
I'll test it as soon as I can, later this month. About to go on vacation without my dev-kit :)

user2684, 2018-10-03:
Thanks, no problem of course and enjoy your vacation!

ttytyper, 2018-10-30:
Tested #417. Oh wow, quite a lot of changes! :) But I like the current structure.

The test was a partial success.

Si7021 + battery works correctly (didn't work before)
Si7021 + signal works correctly (didn't work before)
Battery + signal works correctly
Si7021 + battery + signal gives a small amount of gibberish

Perhaps there's some kind of buffer overflow? I'll do what I can to try to locate the bug, but it will take some time. Is there something else I can do to help you along?

Output from the node, with si7021 + battery + signal enabled:

0 NM:INIT:VER=1.8-dev
0 NM:INIT:INO=NodeManager v1.0
0 NM:INIT:LIB VER=2.3.0-alpha CP=RNNNAA-X 
0 NM:INIT:RBT p=255
2 NM:BFR:INIT
2969 NM:BFR:OK
3246 NM:PRES:BATTERY(201) p=30 t=38
3385 NM:PRES:SIGNAL(202) p=33 t=37
3526 NM:PRES:3665 NM:PRES:	␌�␌-␌���r(r((180) p=8 t=54
4040 NM:STP:ID=1 M=1
4114 NM:STP:HW V=-1 F=8 M=137
4188 NM:LOOP:BATTERY(201):SET t=38 v=3.26
4462 NM:LOOP:(202):SET t=37 v=0

Output from the gateway:

1600 GWT:TPC:CONNECTING...
2102 GWT:TPC:CONNECTING...
2463 TSF:MSG:READ,1-1-255,s=255,c=3,t=7,pt=0,l=0,sg=0:
2468 TSF:MSG:BC
2470 TSF:MSG:FPAR REQ,ID=1
2472 TSF:CKU:OK,FCTRL
2474 TSF:MSG:GWL OK
2909 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=8,pt=1,l=1,sg=0,ft=0,st=OK:0
2915 GWT:TPC:CONNECTING...
3417 GWT:TPC:CONNECTING...
3919 GWT:TPC:CONNECTING...
3921 GWT:TPC:IP=192.168.144.148
3924 MCO:BGN:STP
MY I=0 M=255
INT P=3 M=255
INT P=2 M=255
3926 MCO:BGN:INIT OK,TSP=1
3932 GWT:TPC:IP=192.168.144.148
3935 GWT:RMQ:MQTT RECONNECT
3962 GWT:RMQ:MQTT CONNECTED
3964 GWT:TPS:TOPIC=mysgw-out/0/255/0/0/18,MSG SENT
OK
3969 GWT:TPS:TOPIC=mysgw-out/0/255/3/0/11,MSG SENT
3974 GWT:TPS:TOPIC=mysgw-out/0/255/3/0/12,MSG SENT
READY

4474 TSF:MSG:READ,1-1-0,s=255,c=3,t=24,pt=1,l=1,sg=0:1
4479 TSF:MSG:PINGED,ID=1,HP=1
4491 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=25,pt=1,l=1,sg=0,ft=0,st=OK:1
4497 TSF:MSG:READ,1-1-0,s=255,c=3,t=15,pt=6,l=2,sg=0:0101
4507 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0101
4514 TSF:MSG:READ,1-1-0,s=255,c=3,t=16,pt=0,l=0,sg=0:
4525 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=0,ft=0,st=OK:<NONCE>
4668 TSF:MSG:READ,1-1-0,s=255,c=0,t=17,pt=0,l=11,sg=1:2.3.0-alpha
4675 GWT:TPS:TOPIC=mysgw-out/1/255/0/0/17,MSG SENT
RECV S=1 I=255 C=0 T=17 P=2.3.0-alpha
4681 TSF:MSG:READ,1-1-0,s=255,c=3,t=16,pt=0,l=0,sg=1:
4695 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
4837 TSF:MSG:READ,1-1-0,s=255,c=3,t=6,pt=1,l=1,sg=1:0
4843 GWT:TPS:TOPIC=mysgw-out/1/255/3/0/6,MSG SENT
RECV S=1 I=255 C=3 T=6 P=
5156 GWT:IMQ:TOPIC=mysgw-in/1/255/3/0/6, MSG RECEIVED
5166 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=16,pt=0,l=0,sg=0,ft=0,st=OK:
5288 TSF:MSG:READ,1-1-0,s=255,c=3,t=17,pt=6,l=25,sg=0:<NONCE>
5300 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=6,pt=0,l=1,sg=1,ft=0,st=OK:M
5445 TSF:MSG:READ,1-1-0,s=255,c=3,t=16,pt=0,l=0,sg=1:
5456 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
5599 TSF:MSG:READ,1-1-0,s=255,c=3,t=11,pt=0,l=11,sg=1:NodeManager
5606 GWT:TPS:TOPIC=mysgw-out/1/255/3/0/11,MSG SENT
RECV S=1 I=255 C=3 T=11 P=NodeManager
5611 TSF:MSG:READ,1-1-0,s=255,c=3,t=16,pt=0,l=0,sg=1:
5625 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
5768 TSF:MSG:READ,1-1-0,s=255,c=3,t=12,pt=0,l=3,sg=1:1.0
5774 GWT:TPS:TOPIC=mysgw-out/1/255/3/0/12,MSG SENT
RECV S=1 I=255 C=3 T=12 P=1.0
5780 TSF:MSG:READ,1-1-0,s=201,c=3,t=16,pt=0,l=0,sg=1:
5793 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
5936 TSF:MSG:READ,1-1-0,s=201,c=0,t=30,pt=0,l=7,sg=1:BATTERY
5942 GWT:TPS:TOPIC=mysgw-out/1/201/0/0/30,MSG SENT
RECV S=1 I=201 C=0 T=30 P=BATTERY
5948 TSF:MSG:READ,1-1-0,s=202,c=3,t=16,pt=0,l=0,sg=1:
5961 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
6105 TSF:MSG:READ,1-1-0,s=202,c=0,t=33,pt=0,l=6,sg=1:SIGNAL
6111 GWT:TPS:TOPIC=mysgw-out/1/202/0/0/33,MSG SENT
RECV S=1 I=202 C=0 T=33 P=SIGNAL
6117 TSF:MSG:READ,1-1-0,s=8,c=3,t=16,pt=0,l=0,sg=1:
6130 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
6273 TSF:MSG:READ,1-1-0,s=8,c=0,t=42,pt=0,l=21,sg=1:_��Kw�}�8���z*�mj␒␇␑␄
6280 GWT:TPS:TOPIC=mysgw-out/1/8/0/0/42,MSG SENT
RECV S=1 I=8 C=0 T=42 P=_��Kw�}�8���z*�mj␒␇␑␄
6286 TSF:MSG:READ,1-1-0,s=0,c=3,t=16,pt=0,l=0,sg=1:
6300 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
6443 TSF:MSG:READ,1-1-0,s=0,c=0,t=36,pt=0,l=5,sg=1:b�␆␆
6449 GWT:TPS:TOPIC=mysgw-out/1/0/0/0/36,MSG SENT
RECV S=1 I=0 C=0 T=36 P=b�␆␆
6454 TSF:MSG:READ,1-1-0,s=255,c=3,t=26,pt=1,l=1,sg=1:2
6472 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=16,pt=0,l=0,sg=1,ft=0,st=OK:
6595 TSF:MSG:READ,1-1-0,s=255,c=3,t=17,pt=6,l=25,sg=1:<NONCE>
6607 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=27,pt=1,l=1,sg=1,ft=0,st=OK:1
7423 TSF:MSG:READ,1-1-0,s=255,c=3,t=16,pt=0,l=0,sg=1:
7437 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
7580 TSF:MSG:READ,1-1-0,s=255,c=3,t=0,pt=1,l=1,sg=1:94
7586 GWT:TPS:TOPIC=mysgw-out/1/255/3/0/0,MSG SENT
RECV S=1 I=255 C=3 T=0 P=
7592 TSF:MSG:READ,1-1-0,s=201,c=3,t=16,pt=0,l=0,sg=1:
7604 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
7747 TSF:MSG:READ,1-1-0,s=201,c=1,t=38,pt=7,l=5,sg=1:3.26
7753 GWT:TPS:TOPIC=mysgw-out/1/201/1/0/38,MSG SENT
RECV S=1 I=201 C=1 T=38 P=
7759 TSF:MSG:READ,1-1-0,s=202,c=3,t=16,pt=0,l=0,sg=1:
7772 TSF:MSG:SEND,0-0-1-1,s=255,c=3,t=17,pt=6,l=25,sg=1,ft=0,st=OK:<NONCE>
7915 TSF:MSG:READ,1-1-0,s=202,c=1,t=37,pt=2,l=2,sg=1:0
7921 GWT:TPS:TOPIC=mysgw-out/1/202/1/0/37,MSG SENT
RECV S=1 I=202 C=1 T=37 P=

user2684, 2018-11-01:
Glad to see some progress, thanks. Yeah, looks like a wrong pointer or something like that but it's weird it works with a configuration and it doesn't with another. Only thing I can see in your logs which worries me is the free memory of 137 bytes which is way too low (NM:STP:HW V=-1 F=8 M=137). 
Would you mind trying disabling nodemanager debug (or other NODEMANAGER_* features) so to save some more memory and check if what you get on the gateway is not garbage data? Thanks!

ttytyper, 2018-11-19:
Did some more testing. Disabled all of the NODEMANAGER_* features except _DEBUG and that made the node work correctly. Perhaps it's really just a case of running out of RAM, and not directly related to the si7021 sensor after all? If so, I'm sorry to have taken up your time, and I'm ok with closing the issue.

user2684, 2018-12-01:
Time is invested for good, not taken up, when troubleshooting so no problem at all! This is also telling me memory management is still way from being optimal in some situations....thanks!

PR #417 (2018-12-02): Fix Si7021 library incompatibilities
Using https://github.com/mysensors/MySensorsArduinoExamples/tree/master/libraries/SI7021 library.
Fixes #347
closed 2018-12-01 by user2684