mysensors/NodeManager #349: DHT library not working

horstefan, 2018-05-01:
Enabling DHT11 gives me this errors in console.

sketch/NodeManager.cpp: In member function 'virtual void SensorDHT::onSetup()':
sketch/NodeManager.cpp:1232:29: warning: invalid conversion from 'int' to 'DHT::DHT_MODEL_t' [-fpermissive]
   _dht->setup(_pin,_dht_type);
                             ^
In file included from sketch/NodeManager.h:351:0,
                 from sketch/NodeManager.cpp:5:
/Users/user/Documents/Arduino/libraries/arduino-DHT-master/DHT.h:58:8: note: initializing argument 2 of 'void DHT::setup(uint8_t, DHT::DHT_MODEL_t)'
   void setup(uint8_t pin, DHT_MODEL_t model=AUTO_DETECT);
        ^
sketch/NodeManager.cpp: In member function 'virtual void SensorDHT::onLoop()':
NodeManager.cpp:1238: error: no matching function for call to 'DHT::readSensor(bool)'
   _dht->readSensor(true);
                        ^
sketch/NodeManager.cpp:1238:24: note: candidate is:
In file included from sketch/NodeManager.h:351:0,
                 from sketch/NodeManager.cpp:5:
/Users/user/Documents/Arduino/libraries/arduino-DHT-master/DHT.h:83:8: note: void DHT::readSensor()
   void readSensor();
        ^
/Users/user/Documents/Arduino/libraries/arduino-DHT-master/DHT.h:83:8: note:   candidate expects 0 arguments, 1 provided
sketch/NodeManager.cpp: In member function 'int NodeManager::registerSensor(Sensor*)':
sketch/NodeManager.cpp:3360:43: warning: return-statement with no value, in function returning 'int' [-fpermissive]
   if (sensor->getChildId() > MAX_SENSORS) return;
                                           ^
sketch/NodeManager.cpp: In member function 'bool NodeManager::renameSensor(int, int)':
sketch/NodeManager.cpp:3400:65: warning: return-statement with no value, in function returning 'bool' [-fpermissive]
   if (old_child_id > MAX_SENSORS || new_child_id > MAX_SENSORS) return;
                                                                 ^
exit status 1
no matching function for call to 'DHT::readSensor(bool)'

user2684, 2018-05-02:
Hi, are you sure you are using this library? https://github.com/mysensors/MySensorsArduinoExamples/tree/master/libraries/DHT

Also, please upgrade NodeManager to v1.7, there have been a number of fixes for the DHT sensor. Thanks

user2684, 2018-07-08:
Hi, is this still an issue after using the library pointed above and v1.7? Thanks

Xedecimal, 2019-04-28:
Could we actually switch this library? There are a few libraries out there including the one that has the same author (Mark Ruys) as the supplied one, then some more

https://github.com/markruys/arduino-DHT
https://github.com/adafruit/Adafruit_DHT_Unified
https://github.com/adafruit/DHT-sensor-library

The reason I would suggest this is there's a lot of other libraries and bulk in that examples package that is unrelated to this sensor, so when creating a new PlatformIO project, it's a pain to add the lib_deps entry in the ini to just quickly easily include DHT support.

Normally I'll just copy that DHT folder manually into my deps folder but it could be way easier than that.

I could switch it over to one of the above libraries and make a pull request if you would like ?

user2684, 2019-05-05:
Agree. But I remember a number of issues while using the Adafruit library (as it was the case up to a couple of versions ago) also in combination with other sensors . Cannot remember the exact issue  but #145 and #170 are good candidates. If we believe switching can bring additional value (despite breaking dependencies of previous releases), I'm ok with it as far as the new library will be as solid as the current one. Probably the one from Mark Ruys looks like the best choice in this case.

user2684, 2020-05-04:
Updated documentation referencing https://github.com/markruys/arduino-DHT which is fully compatible with the previously used library (#510)

closed 2020-05-04 by user2684