mysensors/NodeManager #351: Problem with PowerManager

ar84, 2018-05-04:
Hello everybody.

I have the problem that as soon as I activate the point node.setPowerManager (power) the program runs and also a connected DHT22 transmitter reads the data in the given time but does not send anymore after the Power_Pins have been switched off and switched on again.

below you can see my sensor config:

// before
void before() {
  // setup the serial port baud rate
  Serial.begin(MY_BAUD_RATE);
  /*
  * Configure your sensors below
  */
  // report measures of every attached sensors every 10 seconds
  //node.setReportIntervalSeconds(10);
  // report measures of every attached sensors every 15 minutes
  node.setReportIntervalMinutes(15);
  //set the node to sleep in 30 seconds cycles
  //node.setSleepSeconds(10);
  // set the node to sleep in 15 minutes cycles
  node.setSleepMinutes(15);
  // report battery level every 15 minutes
  battery.setReportIntervalMinutes(15);
  battery.setMinVoltage(3.4);
  battery.setMaxVoltage(4.05);  
  
  // set an offset to -1 to a thermistor sensor
  //thermistor.setOffset(-1);
  // change the id of a the first child of a sht21 sensor
  //sht21.children.get(1)->setChildId(5);
  // report only when the analog value is above 40%
  //analog.children.get(1)->setMinThreshold(40);
  //power all the nodes through dedicated pins
  node.setPowerManager(power)
  node.setRetries(3);

wenn ich die node.setPowerManager(power) auskommentiere dann funktioniert alles soweit wie es soll.

Does anyone have an idea and can help me please

user2684, 2018-05-05:
Hi, marking this as a bug and I'll run a test soon to see if I can reproduce the issue before coming back to you. Thanks!

user2684, 2018-08-02:
Hi, this should have been solved with #391, if you can give it a try would be great. Many many things have changed with this PR so have a quick look at the change log there to avoid to get lost.
Let me know if this is actually solved or not. Thanks!

user2684, 2018-11-17:
Closing this for inactivity, feel free to reopen it if the problem is still there with the development version. Thanks!

closed 2018-11-17 by user2684