mysensors/NodeManager #359: Implementing SensorVoltage?

nikolac, 2018-05-11:
I'm using SensorBattery to report my node's battery level, but I also have a small solar panel upstream of the battery that I want to report the voltage of separately. 

After considering using a generic AnalogInput, I realized that the SensorBattery has all of the code I would need. But, it has sendBatteryLevel embedded in it. 

Two thoughts: 1) make the sendBatteryLevel call conditional, based on an exposed configuration
2) Rename SensorBattery to SensorVoltage, remove the sendBatteryLevel calls. Create a new subclass of SensorVoltage called SensorBattery that includes the sendBatteryLevel.

user2684, 2018-05-13:
I'd probably go with 1) by adding an option to allow the user to choose if reporting the voltage, the battery level of both. Adding to 1.8 queue. Thanks!

user2684, 2018-10-14:
Fixed through #428

PR #428 (2018-10-14): Added setSendBatteryLevel() to SensorBattery
if true call sendBatteryLevel() in addition to send the measured voltage back (default: true).

Set to false for e.g. using SensorBattery like a generic "SensorVoltage" (fixes #359)
closed 2018-10-14 by user2684