mysensors/NodeManager #363: SensorMotion constantly wakes from sleep

jasonmhite, 2018-05-14:
I have a motion sensor that I'd like to sleep except to report when triggered, similar to the example. The motion sensor is connected to INT0 (Arduino pin 2) and the sensor itself works as intended, with NodeManager properly reporting the state of the sensor as long as I don't enable sleep.

The problem is, when I enable sleeping the node constantly wakes from sleep, much faster than the sleep interval. I get a constant stream of messages in the debug output along the lines of 

```
4689 MCO:SLP:MS=180000,SMS=1,I1=255,M1=255,I2=0,M2=1
4698 TSF:MSG:SEND,17-17-0-0,s=255,c=3,t=32,pt=5,l=4,sg=0,ft=0,st=OK:500
5206 TSF:TDI:TSL
5206 MCO:SLP:WUP=-2
5210 TSF:TRI:TSB
5212 TSF:MSG:SEND,17-17-0-0,s=255,c=3,t=33,pt=5,l=4,sg=0,ft=0,st=OK:180000
AWAKE
SLEEP 180s

5222 MCO:SLP:MS=180000,SMS=1,I1=255,M1=255,I2=0,M2=1
5230 TSF:MSG:SEND,17-17-0-0,s=255,c=3,t=32,pt=5,l=4,sg=0,ft=0,st=OK:500
5738 TSF:TDI:TSL
5738 MCO:SLP:WUP=-2
5740 TSF:TRI:TSB
5744 TSF:MSG:SEND,17-17-0-0,s=255,c=3,t=33,pt=5,l=4,sg=0,ft=0,st=OK:180000
AWAKE
SLEEP 180s
```

Except these are roughly once per second (very regular, so I don't think it's being tripped by noise), not every 180s. I get this behavior using the example code for a motion sensor, as well as the actual code I wrote that works otherwise when sleeping is disabled (there are other sensors on board). I've tried various settings, like setting the motion interrupt mode etc with no luck, not really sure where to look to debug. I have confirmed with an oscilloscope that the signal from the motion sensor module is nice and steady without any bouncing.

Hardware is a custom board built around an atmega328PB that also has an SI7021, using an NRF24 module as the transmitter, but I have confirmed everything else works as intended as long as sleep mode isn't enabled. I'm using the latest version from Git as of a few days ago.

user2684, 2018-05-17:
HI, I think the problem is related to MySensors library and the way sleep() is handled on the platform you are using. That -2 of `MCO:SLP:WUP=-2` is the library define MY_SLEEP_NOT_POSSIBLE returned by hwSleep() when it is called (see https://github.com/mysensors/MySensors/search?utf8=%E2%9C%93&q=MY_SLEEP_NOT_POSSIBLE&type=). I guess it is compiling with a wrongly architecture which does not support sleep.

user2684, 2018-05-17:
Thanks, let me know please. Also, try to use a different interrupt pin (e.g. pin 3). I really cannot understand why that MY_SLEEP_NOT_POSSIBLE, usually I expect a -1 (end of the sleep cycle) or a positive number (the interrupt pin which triggered the interrupt)

jasonmhite, 2018-05-18:
Looking at the code for `_sleep`, it should have some debug output indicating why sleep wasn't possible, but I don't get any. It's also the fallthrough value in [this](https://github.com/mysensors/MySensors/blob/153850de808023ece9a09b21e847ea546c45239f/core/MySensorsCore.cpp#L630) block of code. I think that's where it's coming from, because I see the output from lines 642 and 644. I don't have an AVR debugger handy, but when I get a chance I will instrument that code and see if I can pick apart what's going on.

I also ordered some sample original 328P chips from Microchip, when they get here I'm gonna try those and see if it's something to do with the newer chips. They are supposed to be compatible, but I've encountered other bugs already so maybe some sort of strange combination of factors is causing this.

user2684, 2018-07-19:
closing since fixed by #383, please reopen it if it will show up again thanks

PR #383 (2018-07-19): fix interrupt sleep
MySensors requires both interrupts or the first interrupt parameter to be defined.
If first interrupt is INTERRUPT_NOT_DEFINE it will not work.

fixes #363
closed 2018-07-19 by user2684