mysensors/NodeManager #373: NodeManager and MySensors openHAB binding: constant reset

wmarkow, 2018-06-19:
Environment:
 * NodeManager installed on Arduino Uno
 * NodeManager configured to be a Serial Gateway
 * NodeManager has enabled **FEATURE_DEBUG** flag
 * Arduino Uno connected to openHAB
 * openHAB has installed MySensors Binding

Result:
 MySensors openHAB Binding constantly reconnects to serial port causing the Arduino Uno to be restarted as well. This looks a bit poor.

This behavior is a side effect of printing to serial port some "bogus" (debug) characters by NodeManager. When NodeManager is configured to be a Serial Gateway, those debug messages doesn't follow the [MySensors Serial API](https://www.mysensors.org/download/serial_api_20).

I think that MySensors openHAB Binding should be resistant to such a bogus messages. I have already reported this issue as tobof/openhab2-addons/issues/116.

Here are the log messages that are printed to serial port by Arduino Uno with NodeManager:
```
NodeManager v1.7
LIB V=2.3.0-alpha R=- E=- T=G A=A S=- B=-
RADIO...0;255;3;0;14;Gateway startup complete.
0;255;0;0;17;2.3.0-alpha
OK
0;255;3;0;11;NodeManager
0;255;3;0;12;1.0
READY 

MY I=0 M=255 
INT P=3 M=255
INT P=2 M=255
```

While I'm good with NodeManager sendings debug messages to serial port, there is only one thing that could be improved in NodeManager. This message looks faulty but it could be easily improved:
```
RADIO...0;255;3;0;14;Gateway startup complete.
```
I believe that a **new line character** is missing after `RADIO...`

user2684, 2018-06-19:
Thanks for reporting this issue. I believe would be a bit harder to make NodeManager following the MySensors Serial API since the structure of the message is completely different and sometimes it is needed to print out something different than just numbers. Workaround would be to turn FEATURE_DEBUG  OFF, as you have probably already done. 
The newline missing is because NodeManager prints "RADIO..." than after the radio communication has been established, prints out an "OK" so to make radio troubleshooting when the MySensors debug is off easier because the OK never comes if there is any communication issue. Makes sense?

user2684, 2018-07-19:
Fixed by #387 by forcing FEATURE_DEBUG to OFF if MY_GATEWAY_SERIAL is defined.
I've noticed NodeManager's debug for a gateway bring also an additional problem: slow down the communication with the controller, especially at low baud rates, preventing a controller supporting smart sleep to respond timely. Hence disabling debug output for a serial gateway looks like the best option for now

wmarkow, 2018-07-20:
Sounds good for now. I can imagine that a lot of debug messages can slow down the communication (and data processing by Arduino), especially at lower baud rates.

user2684, 2018-07-23:
Reopening it since I found a better solution. With #391 debug output is using MySensors' hwDebugPrint() and is no more redirected to the serial port. In this way, the MySensors library automatically convert any debug output in I_LOG_MESSAGE messages when running as a gateway which is fully compatible with any controllers and NodeManager's debug output is not lost

PR #387 (2018-07-19): Disabled NodeManager's debug output when running a serial gateway
This PR force FEATURE_DEBUG to OFF if MY_GATEWAY_SERIAL is defined. Fixes #373 
Added gateway tests for CI