mysensors/NodeManager #378: DHT11 sensor incompatible with SensorPulseMeter

pleschev, 2018-06-27:
Apologies in advance for the newb query, if you could point me in the right direction, it'll be much appreciated.

I'm new to MySensors, I'm setting up my first sensor node with a Nano / RFM69HW / DHT11 & LM393 to measure temp/humidity & the LED pulses from my power meter. I have a Sensebender gateway sending MQTT successfully. Comms between the sensor node & the gateway is fine.

```
#define USE_DHT
#define USE_PULSE_METER
```

```
SensorPowerMeter powerMeter(node,3);
SensorDHT11 dht11(node,4);

```

The behaviour that I'm debugging is that the KWH reported is less than what the power meter is reporting. I found the amount to be off more under higher electricity use. I've dug into the code and found 2 issues:
- https://github.com/mysensors/NodeManager/blob/master/NodeManagerLibrary.ino#L3071 - the count here should be accumulative, as it is currently be reset on every loop send https://github.com/mysensors/NodeManager/blob/master/NodeManagerLibrary.ino#L3052 (this isn't a big deal, I was planning on creating a separate PR for this)
- The root cause of why my reported numbers weren't adding up to the actual power meter is because SensorDHT::onLoop() calls sleepOrWait(1000) - this actually happens twice, once for humidity & again for temperature: https://github.com/mysensors/NodeManager/blob/master/NodeManagerLibrary.ino#L1475 . During this time (when wait(1000) is called), SensorPulseMeter::onInterrupt() doesn't not receive any interrupt calls, thus we miss out on pulses, and are then under-representing the power usage.

My understanding is that interrupts will be lost while in the wait() call. What's the best approach to fix the problem?

user2684, 2018-06-28:
Hi, thanks for reporting the issue. I understand the problem but I'd need some time to reproduce in order to fully understand where the problem is. I'll keep you posted, meanwhile I've marked it as a bug and added to the v1.8 queue. Thanks

pleschev, 2018-06-30:
From what I can tell, during the time sleepOrWait(1000) is called, NodeManager::_onInterrupt_1() does get called, _last_interrupt_pin & co values do get updated, however Sensor::interrupt() and thus SensorPulseMeter::onInterrupt() don't get called.

In terms of reproducing the problem, I have a flashlight app on my phone which flashes n times a second in front of the LM393 which allows me to consistently reproduce the problem.

user2684, 2018-08-02:
Hi, this should have been solved with #391, if you can give it a try would be great. Many many things have changed with this PR so have a quick look at the change log there to avoid to get lost.
Let me know if this is actually solved or not. Thanks!

pleschev, 2019-07-06:
I can confirm this is still a problem with 1.8

user2684, 2019-07-07:
Ok, thanks, this is strange, I will try to reproduce the issue with latest version once the development of the new one starts. Thanks!

user2684, 2020-05-04:
Hi, got to this after a long time apologize but unfortunately couldn't find a solution :-/ Problem is DHT getMinimumSamplingPeriod() is around 2 seconds (so a total of 4 seconds) and I expect multiple interrupts are received during this timeframe. The interrupt is correctly registered however only one is kept for future processing hence the inconsistency you are experiencing. Solution would be keep track of an array of interrupts but the resulting logic would be really complex and would impact other areas of the code. Silly suggestion would be to switch from DHT to other sensors like SHT/SI7021 which do not require waiting/sleeping between measures. If you have thought meanwhile a better approach, feel free to reopen the issue. Thanks!

closed 2020-05-04 by user2684