mysensors/NodeManager #395: Add support for PCA9685 as  RGB/RGBW/W dimmer

sepo83, 2018-07-25:
Add support for PCA9685 (16-channel PWM-board). It should be possible to address the channels as:
* single channel (S_DIMMER)
* RGB (S_RGB_LIGHT)
* RGBW (S_RGBW_LIGHT)

user2684, 2018-07-25:
Adding it to the v1.8 queue, let's see if I can make it. Thanks

sepo83, 2018-07-25:
Sorry, not sure about the workflow here...
I have a working peace of code and  just created a pull-request for this feature #396

user2684, 2018-07-25:
BTW, what a nice addition, lot of work, thanks a lot for investing your time in it!

user2684, 2018-10-13:
Hi, it has been time consuming but I've been finally able to merge your new sensors into the development branch. Since so many things have changed with the architecture review of #391 I did the changes on your behalf re-working your PR #396 (just closed since no more current) into #421. Your code has been entirely ported into the new architecture with the only change of using child of type String instead of int for SensorPca9685Rgb and SensorPca9685Rgbw since looks like this is what you are using. Hope the re-work was correct (also ignore a global bug with getAvailableChildId() fixed in other sensors in that PR). Thanks again for the contribution!

closed 2018-10-13 by user2684