mysensors/NodeManager #400: ESP32 support

knopserl, 2018-08-17:
mysensors has got ESP32 support by tekka007, I was assuming, that also Nodemanager will be ESP32 supported, but it's maintainted seperately.
Since I'm using mysensors with ESP32 barebone boards (for the gateway and also for the sensor nodes), it would be nice to use Nodemanager with ESP32 otherwise I can't use it and would do my own implementation based on the mysensors core.
Since ESP8266 seem to be already supported, I assume, that the ESP32 support should be doable.
I have started to do it by myself, but it would be more efficient, if somebody who knows the mysensors core and nodemanager much better will do it.

user2684, 2018-08-17:
Hi, thanks for your message. I do not have an ESP32 to do a full test with it but generally speaking it should work since NodeManager does not use anything which is hardware specific. If you or somebody else could give it a try and report back would be great, waiting for an ESP32 could take a while.
Nothing special to do, just configure it the same way you are used to with the same directives of your sketch and should compile just fine. Thanks!

knopserl, 2018-08-17:
@user2684 Hi and thanks for your quick feedback.
I wrote this request because I have tried Nodemanager already with ESP32 boards (different board settings) and I got a lot of compile errors and I have tried to fix some of them, but it's very difficult if you don't know Nodemanagers concepts quite well (and of course I did a lot of reaseach of ESP32 HW internals).

First I give you some description what I'm trying to do (the use case) and then I provide a cuple of compile  errors.

I'm trying to set up a quite simple sensor node with temp, humidity and pressure with a battery (LiPo) on an ESP32 with an RFM95 (LoRa), so nothing special I would say.
Thus I wanted also to report battery level (on A0 of ESP32) and RSSI and get the datetime and sleep during reading after reporting the paramerter to save battery.
I use a SHT31 for a more precise temp/humidity and the BME280 only for air pressure.
You might wonder whay I use an ESP32? Well I use an E-Paper for displaying the values and the datetime and e-Paper wanted also an SPI bus (I use one for the RFM95 and one for the E-Paper). The E-Paper wants a lot of RAM (more than usually Arduinos and STM32 offer). The ESP32 hat 520kB RAM, 8k deep sleep (RTC) RAM and 16MB flash and it uses very little sleep power due to the ULP co-processor.
I also wanted to use OTA updates.

Here are the compile errors:
ESP32_NodeManager-SensorNode:325: error: 'SensorConfiguration' does not name a type

 SensorConfiguration configuration(node);

 ^

D:\mcdev\arduino\portable\sketchbook\ESP32_NodeManager-SensorNode\NodeManagerLibrary.ino: In member function 'virtual void SensorBattery::onSetup()':

NodeManagerLibrary:840: error: 'INTERNAL' was not declared in this scope

     analogReference(INTERNAL);

                     ^

NodeManagerLibrary:840: error: 'analogReference' was not declared in this scope

     analogReference(INTERNAL);

                             ^

D:\mcdev\arduino\portable\sketchbook\ESP32_NodeManager-SensorNode\NodeManagerLibrary.ino: In member function 'int SensorAnalogInput::_getAnalogRead()':

NodeManagerLibrary:979: error: 'analogReference' was not declared in this scope

     analogReference(_reference);

                               ^

D:\mcdev\arduino\portable\sketchbook\ESP32_NodeManager-SensorNode\NodeManagerLibrary.ino: In member function 'void NodeManager::reboot()':

NodeManagerLibrary:4976: error: 'WDTCSR' was not declared in this scope

     WDTCSR |= (1<<WDCE) | (1<<WDE);

     ^

NodeManagerLibrary:4976: error: 'WDCE' was not declared in this scope

     WDTCSR |= (1<<WDCE) | (1<<WDE);

                   ^

NodeManagerLibrary:4976: error: 'WDE' was not declared in this scope

     WDTCSR |= (1<<WDCE) | (1<<WDE);

                               ^

D:\mcdev\arduino\portable\sketchbook\ESP32_NodeManager-SensorNode\NodeManagerLibrary.ino: In member function 'float NodeManager::getVcc()':

NodeManagerLibrary:5032: error: 'ADMUX' was not declared in this scope

       ADMUX = (_BV(REFS0) | _BV(MUX3) | _BV(MUX2) | _BV(MUX1));

       ^

In file included from sketch\ESP32_NodeManager-SensorNode.ino.cpp:1:0:

[... 144 lines cut ...]
                 from D:\mcdev\arduino\portable\sketchbook\ESP32_NodeManager-SensorNode\ESP32_NodeManager-SensorNode.ino:316:

D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/core/MySensorsCore.cpp:656:8: note: candidate: int8_t sleep(uint32_t, bool)

 int8_t sleep(const uint32_t sleepingMS, const bool smartSleep)

        ^

In file included from D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32/tools/sdk/include/newlib/stdio.h:29:0,

                 from D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32\cores\esp32/Arduino.h:27,

                 from sketch\ESP32_NodeManager-SensorNode.ino.cpp:1:

D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32/tools/sdk/include/newlib/sys/unistd.h:185:17: note: candidate: unsigned int sleep(unsigned int)

 unsigned _EXFUN(sleep, (unsigned int __seconds ));

                 ^

D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32/tools/sdk/include/newlib/_ansi.h:65:30: note: in definition of macro '_EXFUN'

 #define _EXFUN(name, proto)  name proto

                              ^

knopserl, 2018-08-17:
@user2684 Ok I somehow got it to run to some extent. 
It compiles now without errors and it runs, but the battery sensing does of course not work properly (maybe I have to tell the ESP32 A0 pin or the whole analog read has to be modified).

What I did in in NodeManagerLibrary:
"#undef CHIP_AVR // ESP32  (seem to be define somewhere)"
and comment out the 
;// analogReference(INTERNAL); in line 842

now it looks like:
CONFIG I=200 P=23 T=48
SIGNAL I=202 P=33 T=37
ANALOG_I I=1 P=23 T=48
BME280 I=2 P=6 T=0
BME280 I=3 P=7 T=1
BME280 I=4 P=8 T=4
BME280 I=5 P=8 T=5
SHT31 I=6 P=6 T=0
SHT31 I=7 P=7 T=1
RADIO...37 TSM:INIT
49 TSF:WUR:MS=0
60 TSM:INIT:TSP OK
62 TSM:INIT:STATID=100
64 TSF:SID:OK,ID=100
65 TSM:FPAR
68 TSF:MSG:SEND,100-100-255-255,s=255,c=3,t=7,pt=0,l=0,sg=0,ft=0,st=OK:
711 TSF:MSG:READ,0-0-100,s=255,c=3,t=8,pt=1,l=1,sg=0:0
716 TSF:MSG:FPAR OK,ID=0,D=1
2076 TSM:FPAR:OK
2077 TSM:ID
2078 TSM:ID:OK
2079 TSM:UPL
2174 TSF:MSG:SEND,100-100-0-0,s=255,c=3,t=24,pt=1,l=1,sg=0,ft=0,st=OK:1
2300 TSF:MSG:READ,0-0-100,s=255,c=3,t=25,pt=1,l=1,sg=0:1
2305 TSF:MSG:PONG RECV,HP=1
2307 TSM:UPL:OK
2309 TSM:READY:ID=100,PAR=0,DIS=1
2421 TSF:MSG:SEND,100-100-0-0,s=255,c=4,t=0,pt=6,l=10,sg=0,ft=0,st=OK:FFFFFFFFFFFFFFFF0300
3147 TSF:MSG:SEND,100-100-0-0,s=255,c=3,t=15,pt=6,l=2,sg=0,ft=0,st=OK:0100
3261 TSF:MSG:READ,0-0-100,s=255,c=3,t=15,pt=6,l=2,sg=0:0100
3375 TSF:MSG:SEND,100-100-0-0,s=255,c=0,t=17,pt=0,l=11,sg=0,ft=0,st=OK:2.3.1-alpha
4055 TSF:MSG:SEND,100-100-0-0,s=255,c=3,t=6,pt=1,l=1,sg=0,ft=0,st=OK:0
4187 TSF:MSG:READ,0-0-100,s=255,c=3,t=6,pt=0,l=1,sg=0:M
OK
4306 TSF:MSG:SEND,100-100-0-0,s=255,c=3,t=11,pt=0,l=16,sg=0,ft=0,st=OK:ESP32-SensorNode
5032 TSF:MSG:SEND,100-100-0-0,s=255,c=3,t=12,pt=0,l=3,sg=0,ft=0,st=OK:1.0
PRES I=201 T=30 D=BATTERY
5800 TSF:MSG:SEND,100-100-0-0,s=201,c=0,t=30,pt=0,l=7,sg=0,ft=0,st=OK:BATTERY
PRES I=200 T=23 D=
6503 TSF:MSG:SEND,100-100-0-0,s=200,c=0,t=23,pt=0,l=0,sg=0,ft=0,st=OK:
PRES I=202 T=33 D=SIGNAL
7224 TSF:MSG:SEND,100-100-0-0,s=202,c=0,t=33,pt=0,l=6,sg=0,ft=0,st=OK:SIGNAL
PRES I=1 T=23 D=ANALOG_I
7984 TSF:MSG:SEND,100-100-0-0,s=1,c=0,t=23,pt=0,l=8,sg=0,ft=0,st=OK:ANALOG_I
PRES I=2 T=6 D=BME280
8686 TSF:MSG:SEND,100-100-0-0,s=2,c=0,t=6,pt=0,l=6,sg=0,ft=0,st=OK:BME280
PRES I=3 T=7 D=BME280
9392 TSF:MSG:SEND,100-100-0-0,s=3,c=0,t=7,pt=0,l=6,sg=0,ft=0,st=OK:BME280
PRES I=4 T=8 D=BME280
10104 TSF:MSG:SEND,100-100-0-0,s=4,c=0,t=8,pt=0,l=6,sg=0,ft=0,st=OK:BME280
PRES I=5 T=8 D=BME280
10828 TSF:MSG:SEND,100-100-0-0,s=5,c=0,t=8,pt=0,l=6,sg=0,ft=0,st=OK:BME280
PRES I=6 T=6 D=SHT31
11576 TSF:MSG:SEND,100-100-0-0,s=6,c=0,t=6,pt=0,l=5,sg=0,ft=0,st=OK:SHT31
PRES I=7 T=7 D=SHT31
12272 TSF:MSG:SEND,100-100-0-0,s=7,c=0,t=7,pt=0,l=5,sg=0,ft=0,st=OK:SHT31
READY

knopserl, 2018-08-18:
I did some further investigation and there are some HW related topics which requires some work.
I made some enhancements In the NodeManagerLibrary.h  such as 
#if defined(ESP32) || defined(MY_GATEWAY_ESP32)
  #define CHIP_ESP32
 ... 
#ifdef CHIP_ESP32
  #include <WiFi.h>
#endif
.... 
#if !defined(CHIP_ESP8266) && !defined(CHIP_ESP32) && !defined(CHIP_STM32) && !defined(CHIP_NRF5)
  #define CHIP_AVR
#endif

to get the internal ESP32 VCC values with getVccc (ESP.getVcc) there must be this definition outside the code, but I need to find out which header file I need for this
ADC_MODE(ADC_VCC) 

Using one of the 12bit ADC's the Battery voltage feature works but requires some more calibration (I got a 3.4V, 94% where I should have 3,7V and 100%).

The biggest open topic seem to be the sleep mode to save battery, either the light sleep with 0,8mA or better the deep sleep with 6-10nA, where the ESP does a warm start after wakeup by it's internal timer (runs through the init/setup code again, but can read a saved state from it's internal RTC RAM).
For me it looks like the sleep mode has not yet been adaopted for the ESP's (ESP8266, ESP32), but I need to ask tekka who did a ESP32 support for mySensors.

user2684, 2018-08-20:
Thanks for the detailed explanation! If it is a compile issue I can run some tests even without the device, thanks for sharing with me all the details and the fix. I should pay more attention to keep out of the code Arduino-specific variables. 
I'll review your proposed solution and merged directly into #391.
Regarding the sleep mode, unfortunately looks like it is not supported by the library (https://github.com/mysensors/MySensors/blob/development/hal/architecture/ESP32/MyHwESP32.cpp#L90). Thanks again!

knopserl, 2018-08-20:
@user2684 welcome. I got Nodemanager actually quite well running with my two supported sensors (SHT31 and BME280). Also the battery sensor with once of the (12) ADC pins works (some calibration is required), but that is not a problem.
The biggest issue (and that is not within NodeManager) is the power save mode support for ESP chips (ESp8266 or ESP32) although both and especially the ESP32 hast advanced power save HW features and API's but it is not yet implemented in the MySensors core. The sleep function returns immediatly with false and it does not sleep. I got this info from @mfalkvidd. SO this part is not yet implemented. If I find some time, I'll try to work on a sleep feature but I'm not sure if it breaks somehow the concept as the ESP has to be set in one of the deep_sleep modes (to really come doen to a few uA) and then it does more or less a restart/reboot but alows to keep 8k of data in it's static RTC RAM.
That means it going through the setup code again and you have to decide what kind of tasks shall be done as part of a cold start and what during the wakeup warm start. I don't know how this can fit to the mysensors loop concept. 
You might have some hints, e.g. registration (presentation) can be skipped, but sensors and RFM transciver need to be reinitialized as the also a set to power down during ESP deep sleep.

user2684, 2018-10-01:
@knopserl Hi, would you mind giving a try to #414? Unfortunately I have no hardware for an intensive test.
Also keep in mind NodeManager's architecture has changed quite a lot in this development release so we may want to give a look at #391 if you get lost. Thanks!

knopserl, 2018-10-02:
@user2684 Yes I'll do so and let you know.
Thanks for yor patience and support. I like the mySensors and NodeManger project but I have decided to move away from the Arduino/Atmega MCU's towards to ESP MCU's (mainly ESP32) for not-battery powered devices (because project are not really supporting the deep sleep concept of the ESP's) and I'm starting with nRF52 (832, 840) ARM MCU's for battery-powered devices which are more engergy efficient, have radio on-chip and all (ESP and nRF) have magnitues more Flash and RAM.
Ransery was so kind and made very noce PCB's for the ESP32 and nRF52 modules compatible also for MySensors projects.
I don't expect the MySensors developer to support ever ESP deep sleep (even it is not bad but different) but hope we get full nRF52 support (I have to find out what is missing).

knopserl, 2018-10-02:
@user2684 I took the Development branch (is that what you want me to test)?
So I took the Development branch and coiped it to the libraries and since my last working ESP32 Nodemanger does no longer work, I took your exaples/template.ino and just seleted a few settings like the RFM95 and the battery, analogread and the signal features (uncommended the lines)

However I get a lot of compile errors. I also took the latest MySensors Development libraries afterwards because I though my existing ones (2.3 release) does no longer match. But the same result.
I think I had such a list of errors already but I can't remember what I did. I thought is again the #undef CHIP_AVR I had to add, but it did not solve it. I guess there preprosessore still think I have an AVR chip and not an ESP32.
So if you're saying you have compiled it with an chosen ESP32 board (not having one connected), what board have you chosen? I have tested at least 4 different ESP32 boards, although I have a bare ESP-WROOM-32, also called ESP32S mounted on a Mysensors board from ranseyer). It has already compiled and did run with my adapted code long time ago. 

Compile errors:
In file included from D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/MySensors_NodeManager.h:82:0,
                 from D:\mcdev\arduino\portable\sketchbook\_ESP32\ESP32_NodeManager_Template\ESP32_NodeManager_Template.ino:178:
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp: In member function 'void NodeManager::reboot()':
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp:356:4: error: 'WDTCSR' was not declared in this scope
    WDTCSR |= (1<<WDCE) | (1<<WDE);
    ^
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp:356:18: error: 'WDCE' was not declared in this scope
    WDTCSR |= (1<<WDCE) | (1<<WDE);
                  ^
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp:356:30: error: 'WDE' was not declared in this scope
    WDTCSR |= (1<<WDCE) | (1<<WDE);
                              ^
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp: In member function 'void NodeManager::setADCOff()':
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp:449:3: error: 'ADCSRA' was not declared in this scope
   ADCSRA = ADCSRA & B01111111;
   ^
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp:451:3: error: 'ACSR' was not declared in this scope
   ACSR = B10000000;
   ^
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp: In member function 'void NodeManager::sleepOrWait(long unsigned int)':
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp:458:69: error: call of overloaded 'sleep(long unsigned int&)' is ambiguous
   if (isSleepingNode() && _sleep_or_wait && value > 200) sleep(value);
                                                                    ^
In file included from D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/MySensors.h:433:0,
                 from D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/MySensors_NodeManager.h:43,
                 from D:\mcdev\arduino\portable\sketchbook\_ESP32\ESP32_NodeManager_Template\ESP32_NodeManager_Template.ino:178:
D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/core/MySensorsCore.cpp:666:8: note: candidate: int8_t sleep(uint32_t, bool)
 int8_t sleep(const uint32_t sleepingMS, const bool smartSleep)
        ^
In file included from D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32/tools/sdk/include/newlib/stdio.h:29:0,
                 from D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32\cores\esp32/Arduino.h:27,
                 from sketch\ESP32_NodeManager_Template.ino.cpp:1:
D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32/tools/sdk/include/newlib/sys/unistd.h:185:17: note: candidate: unsigned int sleep(unsigned int)
 unsigned _EXFUN(sleep, (unsigned int __seconds ));
                 ^
D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32/tools/sdk/include/newlib/_ansi.h:65:30: note: in definition of macro '_EXFUN'
 #define _EXFUN(name, proto)  name proto
                              ^
In file included from D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/MySensors_NodeManager.h:82:0,
                 from D:\mcdev\arduino\portable\sketchbook\_ESP32\ESP32_NodeManager_Template\ESP32_NodeManager_Template.ino:178:
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp: In member function 'void NodeManager::setAnalogReference(uint8_t, uint8_t)':
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp:731:23: error: 'analogReference' was not declared in this scope
  analogReference(value);
                       ^
In file included from D:\mcdev\arduino\portable\sketchbook\_ESP32\ESP32_NodeManager_Template\ESP32_NodeManager_Template.ino:186:0:
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/sensors/SensorBattery.h: In member function 'virtual void SensorBattery::onLoop(Child*)':
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/sensors/SensorBattery.h:85:35: error: 'INTERNAL' was not declared in this scope
    nodeManager.setAnalogReference(INTERNAL);
                                   ^

user2684, 2018-10-03:
Thanks for testing it out so quickly! Sorry,my bad, I meant if you can test PR #414 (https://github.com/user2684/NodeManager/tree/arch/esp32) which is not yet into the development version. This is where I applied the patch. At least here it compiles but I'm not very confident without a decent test. And about what you said in the previous message, I fully understand your point, deep sleep is a fundamental requirement :-) Even if at least when I tested a bunch of ESP8266, still their power consumption when awake was far too much for lasting more than a few days on batteries but I am far from being an expert with this stuff!

knopserl, 2018-10-03:
@user2684 ok, I did not know how to get the branch for PR #414 (still don't know how to link the pull request # to the branch) but with your link to the branch I did the dowaload, installed the lib and took the template example from this branch and still get compile errors (but less). Obviously something with the sleep() does not match (with the mysensors library)?

In file included from D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/MySensors_NodeManager.h:82:0,
                 from D:\mcdev\arduino\portable\sketchbook\_ESP32\ESP32_NodeManager_Template\ESP32_NodeManager_Template.ino:176:
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp: In member function 'void NodeManager::sleepOrWait(long unsigned int)':
D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp:458:69: error: call of overloaded 'sleep(long unsigned int&)' is ambiguous
   if (isSleepingNode() && _sleep_or_wait && value > 200) sleep(value);
                                                                     ^
In file included from D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/MySensors.h:433:0,
                 from D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/MySensors_NodeManager.h:43,
                 from D:\mcdev\arduino\portable\sketchbook\_ESP32\ESP32_NodeManager_Template\ESP32_NodeManager_Template.ino:176:
D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/core/MySensorsCore.cpp:666:8: note: candidate: int8_t sleep(uint32_t, bool)
 int8_t sleep(const uint32_t sleepingMS, const bool smartSleep)
        ^
In file included from D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32/tools/sdk/include/newlib/stdio.h:29:0,
                 from D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32\cores\esp32/Arduino.h:27,
                 from sketch\ESP32_NodeManager_Template.ino.cpp:1:
D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32/tools/sdk/include/newlib/sys/unistd.h:185:17: note: candidate: unsigned int sleep(unsigned int)
 unsigned _EXFUN(sleep, (unsigned int __seconds ));
                 ^
D:\mcdev\arduino\portable\sketchbook\hardware\espressif\esp32/tools/sdk/include/newlib/_ansi.h:65:30: note: in definition of macro '_EXFUN'
 #define _EXFUN(name, proto)  name proto

BTW: I knwo from MySensors, that the ESP32 ist supported and does not show compile errors, but the sleep does not stop/sleep but just continues an returns an error code (not supported) which can be ignored.
Fot MySensors Gateways, the ESP32 works perfect, even ESP32 with RFM95 on-board. I use it with that configuration.
But it would be nice, if Nodemanager would compile and work without sleep for all other use cases.

What I'm now trying is to get Mysensors/Node Manager running on my nRF52 boards which are really fit 100% to battery powered projects. Lets see what errors I get then (nRF52 should be supportd in principle) and have the expected sleep/wakeup behavior (keeps its state, does not a restart etc.)

user2684, 2018-10-05:
Thanks a lot for your tests, this was exactly what I needed. It doesn't look like an issue with sleep (as you said, when not implemented, should still compile just fine) but more an inability to cast between data types. I still need to understand why since looks pretty strange. If meanwhile you can briefly let me know how to configure the Arduino IDE to run a test similar to yours would be great because the way I set it up here works fine so there must be something wrong I'm doing and I do not want to overload you with requests of testing every single little change. Many thanks in advance!

knopserl, 2018-10-05:
@user2684 I have tried different nRF boards, but it does not ake a difference.
Here is my Arduino settings (screenshot). Let me know if you need more/other inos.

[image]

knopserl, 2018-10-06:
@user2684 your're welcome and I will try to support you as best I can.
In principle your should be fine with your ESP32 setup. Theoretically, it could be also my Arduino setup. HOwever - as I mentioned - the MySensor ESP32 scetches compiling and running fine (except the non implemented sleep)
So I have removed all my ESP32 board hardware folder and will do a fresh install and give it another try and let you know if something has changed (after I got some hours sleep).
I have a portable Arduino setup, but that should make no difference.
You could try with two other ESP32 board settings.
I use very often ESP32 bare boards (ESP32S, Wroom32)  so I choose ESP32 Dev Module. Or you can also try to use WEMOS LOLIN32 (if it is in your list). These are very common used ESP32 boards. The might come with different compiler settings. I get errors with both.

Here is my preferences setting (for the board manager)

https://dl.espressif.com/dl/package_esp32_index.json
http://arduino.esp8266.com/stable/package_esp8266com_index.json
http://dan.drown.org/stm32duino/package_STM32duino_index.json
https://www.adafruit.com/package_adafruit_index.json
https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json
https://raw.githubusercontent.com/mysensors/ArduinoBoards/master/package_mysensors.org_index.json

knopserl, 2018-10-07:
@user2684 I have tested a bit further and came to the following findings

there are multiple confliction sleep definitions, the compiler can't resolve (unabiguous) for the defintion of 
sleep (SLEEP_TIME) or sleep (unsigned int)
in
MySensors/core/MySensorsCore.cpp:666:8: note: candidate: int8_t sleep(uint32_t, bool)
and 
arduino\portable\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\sys\unistd.h:185:10: note: candidate: unsigned int sleep(unsigned int)
 unsigned _EXFUN(sleep, (unsigned int __seconds ));

I don't know, why there are so many different definitions of sleep at differnet places.
If I use sleep(SLEEP_TIME, true); the problem is gone, if I use sleep(SLEEP_TIME) then there are multiple definitions.

I have so far not understood the recommened folder structure for HW defintions as it is installed by the IDE without asking. There is a HW dfintion in <sketchdir>\hardware and defintions are also in portable\packages. If remove it in the one or other place, the IDE does not start or throws errors.

user2684, 2018-10-12:
@knopserl I've been finally able to reproduce your issue and hopefully fixed it. If you can download again https://github.com/user2684/NodeManager/tree/arch/esp32 and give it a try would be great! 
Let's move now to nRF52832 support. I have understood https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json is the URL to use for the boards but I see you reference a "MyBoard-something" in your code. Could you please share some sort of step by step instructions (board to use, file to include, define to use, etc.) so I can reproduce the issue? Thanks!

knopserl, 2018-10-12:
@user2684 I have tested your new versionand it really looks good, very good job. 
It compiles without any warnings or errors and it runs also without errors.
I'M using the ESP32 Dev Module Board in the Board settings as I have a bare ESP32 module (see picture below) made for me by ranseyer (see openhardware.io)
https://www.openhardware.io/view/624/ESP32-MYSX-Baord (the typo in the link is correct: baord)
I'm using a RFM95 (LoRa) radio and it also seem to work.
The only thing which seem not to work (could be my fault) is the battery level.
I'm using A6 (pin 34) and I have tried w/ and w/o analog sensor (check the values).
It says I have 0.0 Volt. I have the standard voltage devider form MySensors on the analog input (usually A0, I have connected it on A6) with 470k/1M Ohm und I measure 0,98V which shall report 3.3V and 100%. In the run below you see, that I used both the battery(A6) and the AnalogSensor (A6). Even if I don't use analog sensor, I get 0.0 V. Maybe I'm doing something wrong or I have to define the reference Voltage somewhere?

I will respond to the nRF52 in a seperate post!

#include <sensors/SensorBattery.h>
//SensorBattery battery(A0);
SensorBattery battery(A6);

//#include <sensors/SensorConfiguration.h>
//SensorConfiguration configuration;

#include <sensors/SensorSignal.h>
SensorSignal signal;

#include <sensors/SensorAnalogInput.h>
SensorAnalogInput analog(A6);

Here is the proof (excerpt of the logfile)
4272 MCO:PIM:NODE REG=1
4274 MCO:BGN:STP
4275 NM:STP:ID=101 M=0
4277 NM:STP:HW V=0 F=240 M=26308
4280 MCO:BGN:INIT OK,TSP=1
4283 NM:LOOP:BATTERY(34):SET t=38 v=0.00
4429 TSF:MSG:SEND,101-101-0-0,s=255,c=3,t=0,pt=1,l=1,sg=0,ft=0,st=OK:0
4435 NM:MSG:SEND(34) t=38 p=0.00
4587 TSF:MSG:SEND,101-101-0-0,s=34,c=1,t=38,pt=7,l=5,sg=0,ft=0,st=OK:0.00
4594 NM:LOOP:SIGNAL(202):SET t=37 v=-39
4745 TSF:MSG:SEND,101-101-0-0,s=202,c=1,t=37,pt=4,l=4,sg=0,ft=0,st=OK:-39
4752 NM:LOOP:ANALOG_I(1):SET t=48 v=84
4903 TSF:MSG:SEND,101-101-0-0,s=1,c=1,t=48,pt=4,l=4,sg=0,ft=0,st=OK:84
4909 NM:SLP:SLEEP s=30

knopserl, 2018-10-12:
@user2684 Now to the secon d interesting topic, the support for the nRF52832:
I have trried a few things like the board manager extension from sandeep mistry
This is what you have to include in the Arduino Preferneces - new board manager URL's:
https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json
And then you have to restart Arduino and go to Tools - Board Manager and it loads the new boards wheer you can pick one e.g. Generic nRF5, but non of them help really (you get a lot of compile errors in MySensors/NodeManger).
Then I got the hint to use this board extension made for MySensors:
https://raw.githubusercontent.com/mysensors/ArduinoBoards/master/package_mysensors.org_index.json
Some procedure as above but you get additional two boards (MyNRF5 nRF52832 ...)
Choose the nRF52832 board.
https://raw.githubusercontent.com/mysensors/ArduinoBoards/master/package_mysensors.org_index.json 

When I then try to compile the Nodemanager I get the following errors:

in file included from D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/MySensors_NodeManager.h:82:0,
                 from D:\mcdev\arduino\portable\sketchbook\_nRF52\nRF52_NodeManager_Template\nRF52_NodeManager_Template.ino:176:

D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp: In member function 'void NodeManager::sendMessage(uint8_t, uint8_t, int)':

D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/nodemanager/Node.cpp:529:21: error: call of overloaded 'set(int&)' is ambiguous
   _message.set(value);
                     ^

In file included from D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/MySensors.h:431:0,
                 from D:\mcdev\arduino\portable\sketchbook\libraries\NodeManager/MySensors_NodeManager.h:43,

                 from D:\mcdev\arduino\portable\sketchbook\_nRF52\nRF52_NodeManager_Template\nRF52_NodeManager_Template.ino:176:

D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/core/MyMessage.cpp:268:12: note: candidate: MyMessage& MyMessage::set(bool)

 MyMessage& MyMessage::set(const bool value)
            ^

D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/core/MyMessage.cpp:276:12: note: candidate: MyMessage& MyMessage::set(uint8_t)

 MyMessage& MyMessage::set(const uint8_t value)

            ^

D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/core/MyMessage.cpp:293:12: note: candidate: MyMessage& MyMessage::set(uint32_t)

 MyMessage& MyMessage::set(const uint32_t value)

            ^

D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/core/MyMessage.cpp:301:12: note: candidate: MyMessage& MyMessage::set(int32_t)

 MyMessage& MyMessage::set(const int32_t value)

            ^

D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/core/MyMessage.cpp:309:12: note: candidate: MyMessage& MyMessage::set(uint16_t)

 MyMessage& MyMessage::set(const uint16_t value)

            ^

D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/core/MyMessage.cpp:317:12: note: candidate: MyMessage& MyMessage::set(int16_t)

 MyMessage& MyMessage::set(const int16_t value)

            ^

user2684, 2018-10-13:
Glad to hear we are making some steps ahead! As for the battery voltage, SensorBattery does not take a pin as the first argument hence it is calling hwCPUVoltage() which returns FUNCTION_NOT_SUPPORTED on ESP32 (https://github.com/mysensors/MySensors/blob/6ffe29cb5fcfa33fa83133044c2d16ee45c66697/hal/architecture/ESP32/MyHwESP32.cpp#L119). If you want to use the analog pin you have to add:
```
battery.setBatteryInternalVcc(false);
battery.setBatteryPin(A6);
```
Let me know if with this would work better. Thanks

knopserl, 2018-10-13:
@user2684 The ESP32 looks good so far. I modified the battery settings and it returns a value which is not correct but it needs just a correct reference volatge (e.g. in my case around 0.9V). Maybe you can give me a another hint how/where to set it, so that I can calibrate the analog reading (12 Bit ADC, 0.9V)

regarding the nRF52: I did configured one instance as a Node and the other as a SerialGateway to test some communication (nRF24 Radio mode, which is supported by the nRF52).

I could compile the Sensor Node with few warings, but could not compile a SerialGateway setting for a second nRF52 node.

Since the compile logs are longer, I attach the two files rather to include them directly in the post.

[Nodemanager_Node_Warnings.txt](https://github.com/mysensors/NodeManager/files/2475497/Nodemanager_Node_Warnings.txt)
[Nodemanager_Serial-Gateway_Errors.txt](https://github.com/mysensors/NodeManager/files/2475498/Nodemanager_Serial-Gateway_Errors.txt)

user2684, 2018-10-14:
Good to know looks like it is working fine. I'm now merging PR #414 since strictly related to ESP32 support and we can continue the discussion of the other issue. Regarding the node warning, nothing to be worried about. Regarding the gateway errors, would you mind sharing also the sketch you are using for that test?
As for the battery settings, I guess you can play with setBatteryVoltsPerBit() of SensorBattery to adjust the calculated value. Thanks!

knopserl, 2018-10-14:
@user2684 no problem sharing the two sketches (nRF52_Nodemanager_Node and Gateway). There is not much I did (mostly Nodemanager standard)

[nRF52_NodeManager_Node_Template.zip](https://github.com/mysensors/NodeManager/files/2476851/nRF52_NodeManager_Node_Template.zip)
[nRF52_NodeManager_Serial_Gateway_Template.zip](https://github.com/mysensors/NodeManager/files/2476853/nRF52_NodeManager_Serial_Gateway_Template.zip)

knopserl, 2018-10-21:
@user2684 thanks! I've tries both, the Node Template and  the Serial Gateway Template.
They compile now file but after start, I get in both sketches a segment fault.
Since the first lines in the sketch is the serial line print "Nodemanager" in the setup block and it does not reach even this line, there seem to be a problem even before that line.
I'll try to do some debugging, but that could be tricky and time consuming. I have'nt done it before using the black magic probe. 

Temporary breakpoint 1 at 0x3a88: file D:\mcdev\arduino\portable\sketchbook\libraries\MySensors/hal/architecture/NRF5/MyMainNRF5.cpp, line 23.
Starting program: C:\Users\internet\AppData\Local\Temp\arduino_build_893561\nRF52_NodeManager_Serial_Gateway_Template.ino.elf 
Note: automatically using hardware breakpoints for read-only addresses.

Program received signal SIGSEGV, Segmentation fault.
0x00000000 in __isr_vector ()

kainhofer, 2018-10-21:
I'm running into the same issues (the latest version does not run at all on my nrf51822 devices). I haven't had time to track down the issue. As a quick test, I commented out all nodemanager-related lines and the setup and loop functions were properly called and debug output was printed to the serial console. Also, the previous release of NodeManager also ran fine on my nrf5 devices.

So it is really some change in the latest devel version of NM. My latest patch mentioned by @user2684 in the previous post does NOT address these issues. Rather they are a forward port of minor changes that I made to the latest stable version.

knopserl, 2018-10-28:
@kainhofer very good job! Maybe @user2684 finds a solution for that. Would be interesting if this is only a problem for the nRF51/52 MCU's and how this can be solved.
Meanwhile, it's no problem to do the serial init in the setup(). How was ist actually done with the nRF51? I use the nRF52832/840. How did you define the serial port pins (as there is no predefined UART's - can be almost at any port)?

user2684, 2018-11-01:
Good finding @kainhofer, I'll review how this is normally handled and apply the required changes to make it work out of the box. Thanks!

user2684, 2018-11-17:
I couldn't find anywhere a good example to initialize the serial port on this board and not having a sample with me doesn't help :-/
As a temporary workaround in #438 I just set the following to at least prevent the board from crashing:
```
#ifndef CHIP_NRF5
	MY_SERIALDEVICE.begin(MY_BAUD_RATE);
#endif
```
Still by initializing the serial port in setup() a few debug messages are lost but better than an unexpected crash I believe. If anybody has a better approach to recommend, please feel free to add a comment in #438 
Thanks!

PR #414 (2018-10-14): ESP32 Support
fixes #400
closed 2018-11-17 by user2684