mysensors/NodeManager #401: NodeManager v1.7 not compiling with Arduino IDE 1.8.6

ChrisWharton2, 2018-09-09:
I recently moved to a new computer and had to setup my environment again, in doing this I upgraded the Arduino IDE to version 1.8.6. My Nodemanager v1.7 sketches would no longer compile and gave a type error on one of the MySensors libraries (Version 2.2).  This problem also occurred on the default NodeManager sketches without modification.
Other sketches using just MySensors examples compiled correctly.
I replicated the issue on another Machine on LInux my main machine is WIndows.
Downgrading the IDE to V1.8.5  resolved the compilation error.  Obvioulsy a problem with the IDE but I suspect you guys are going to have to address.

user2684, 2018-09-09:
Thanks for reporting this issue, I'll run a test and report back. Meanwhile I've marked this as a bug and added to the v1.8 development queue

user2684, 2018-10-01:
Hi, tested with IDE 1.8.7 and it seems to compile just fine. I'm using NodeManager's v1.8-dev, maybe it could be the new version or the new version of the IDE, who knows. Can you please run the same test to check it out? Also would be great if you can share the error when compiling your sketch. Thanks!

ChrisWharton2, 2018-10-02:
ok, have just installed the latest Arduino IDE (1.8.7) and get the same error as with v1.8.6
here is the error message, I do not think this related to Onedrive as I get the same error on linux and using a standard drive. Also the strange pathing with forward and back slashes is not in the linux error this seems to be a problem with the way the paths are reported on windows.

` 
Build options changed, rebuilding all
test_11:447:20: error: 'MyMessage' does not name a type

 void receive(const MyMessage &message) {

                    ^

In file included from C:\Users\chris\OneDrive\App\Arduino\libraries\MySensors/MySensors.h:371:0,

                 from sketch\NodeManagerLibrary.h:2,

                 from C:\Users\chris\OneDrive\App\Arduino\test_11\test_11.ino:312:

C:\Users\chris\OneDrive\App\Arduino\libraries\MySensors/core/MyTransport.cpp: In function 'void transportProcessMessage()':

C:\Users\chris\OneDrive\App\Arduino\libraries\MySensors/core/MyTransport.cpp:845:14: error: cannot resolve overloaded function 'receive' based on conversion to type 'bool'

   if (receive) {

              ^

C:\Users\chris\OneDrive\App\Arduino\libraries\MySensors/core/MyTransport.cpp:911:15: error: cannot resolve overloaded function 'receive' based on conversion to type 'bool'

    if (receive) {

               ^

exit status 1
'MyMessage' does not name a type`

Regards
Chris

user2684, 2018-10-03:
Thanks for sharing the error. Apparently it is not strictly NodeManager-related but anyhow looks strange. I found similar issues in the forum but very old apparently without a solution:
* https://forum.mysensors.org/post/25729
* https://forum.mysensors.org/post/61457
Would you mind compiling a different sketch, without NodeManager but with that version of the IDE, that operating system, that location of the file and that MySensor library version? Thanks

rspaargaren, 2018-10-15:
Hi I have currently the same problem. If I compile a node manager sketch with 1.8.7. I will try the dev version of node manager. If I take the receive part out the node manager does compile.

rspaargaren, 2018-10-16:
Today I have tested two setups:
Arduino 1.8.5 -> Nodemanager 1.7 compiles
Arduino 1.8.7 -> Mysensor dimmer sketch with receive of Mymessage -> compiles
I did not try the dev part of Nodemanager as I did not know what to compile. It is a complete new setup..?

user2684, 2018-10-21:
Thanks! Yeah the dev release is quite different and runs as a standard arduino's library. But if you are saying now it is compiling, I'm good with it. Can I close this issue?

rspaargaren, 2018-10-22:
No the standard mysensor dimmer sketch with the mymessage in it compiles. The nodemanager sketch does not. Same behaviour on compiling on a Ubuntu machine.

rspaargaren, 2018-11-01:
Yes i can try but as it was completely restructured I was not sure what to compile. Can you give me a short instruction and then I will try

user2684, 2018-11-03:
Sure no problem and thanks for your availability. 

* Download the dev release from https://github.com/mysensors/NodeManager/archive/development.zip
* Move the directory in the arduini libraries directory like you are used to do with any other arduino library - https://www.arduino.cc/en/Guide/Libraries)
* Open /restart the arduino IDE, go to Examples and you should find Nodemanager there as well. Select template. 
* Just compile it or do the changes you want and compile

Thanks, I've tried again and it is still compiling fine in my tests so please let me know about your findings.

user2684, 2018-11-11:
Great to hear dev release (which I hope should be released shortly) is comping just fine! Closing the issue. Thanks for your tests!

closed 2018-11-11 by user2684