mysensors/NodeManager #420: Issue with FEATURE_CONDITIONAL_REPORT with interrupt

Bottesford, 2018-10-10:
I have a battery powered node with a BH1750 sensor for which I have set conditional reporting as follows:

`light.children.get(1)->setValueDelta(2);
  light.children.get(1)->setForceUpdateMinutes(25);`

Attached to the same node I have a SensorDoor using interrupts. I had a problem when only rising changes were being sent back to the controller. The interrupt would fire and pre-post sleep messages would be sent but the actual payload:0 message would never be sent. If FEATURE_CONDITIONAL_REPORT was disabled it would work properly.

I tracked it down to _min_threshold var being set to FLT_MIN which comes up as 0. This meant the 0 value was being filtered out in the ChildInt::sendValue(bool force) method.
I set this in my node code:
`door.children.get(1)->setMinThreshold(-FLT_MIN);`
This resolves the issue. 
But probably that was not how this function was intended. Aware you are mid-way through a rewrite so maybe this doesn't apply any more but thought I'd point it out anyway as took some tracking down!

user2684, 2018-10-11:
Thanks for reporting this issue, I've tagged as a bug and I'll try to reproduce it asap.

user2684, 2018-10-14:
You're fully right, looks like FLT_MIN is the minimum (positive) float number. 
With #430 I've set it to -FLT_MAX (-FLT_MIN is always close to zero). 
I've also fixed a few other issues which came out while testing conditional reporting. 
Thanks!

PR #430 (2018-10-14): Fix conditional report issues
This PR fixes the following:

* `_min_threshold` default is now `-FT_MAX` (and no more `FT_MIN ` which is zero) (fixes #420)
* Always send out the value if it is the first run (otherwise comparison is done with 0 which is not always correct)
* Fixed `NodeManager::_sendMessage()` to print debug message when handling long int
closed 2018-10-14 by user2684