mysensors/NodeManager #432: Display update of SSD1306 broken in latest version

kainhofer, 2018-10-15:
In the latest development version, a SSD1306 display will NOT update its texts. The reason is that in Sensor::loop the check 
    	if (_evaluateTimer(_measure_timer) || message != nullptr) {
will never return true, because the timer is explicitly set not to trigger at all in the OLED display's constructor. So by default, the display will update once (when _first_run == false) and then never again.

Since Sensor::loop is not virtual, I also don't see any chance in a Display object to override the loop behaviour, either.

In previous versions, I could add a Child to cause the check in the loop to evaluate to true. However, this has changed in the latest development version and an existing child no longer forces the onLoop function to be called at all. This makes sense, but it breaks the workaround I had to use to make the OLED update so far.

Any idea how to make the OLED displays update, while not sending data to the gateway for them?

user2684, 2018-10-21:
I believe the "obscure" and not documented way this was supposed to work in my mind when restructuring the code was to have the report timer to off  with `setReportTimerMode(DO_NOT_REPORT) ` (a display should never report back to the gateway otherwise would generate unnecessary traffic) but having the user customizing the refresh interval with the measure timer (by calling `setMeasureTimerMode()` and `setMeasureTimerValue()`) through which the display's `onLoop()` is called. 
Makes sense? (if it does I definitely need to document this better...)

kainhofer, 2018-10-21:
Ah, I see. That certainly makes sense. Unfortunately, the logic in Sensor::setup messes up the measure timer for the Display class:
  * Display::Display calls setReportTimerMode(DO_NOT_REPORT)
  * Sensor::setup copies DO_NOT_REPORT to the measure timer, too

Unfortunately, the Sensor::setup function is not virtual, so I don't see any easy way in the Display class to check whether a report interval was explicitly set by the user or implicitly by Sensor::setup...

In my view, the Display class should set the nodeManager's report interval as default for its _measure_timer. Then this issue would be solved.

user2684, 2018-10-21:
Hi, theoretically, measure timer is set to report timer only if the measure timer is not set. So if in your sketch in before() you call e.g. `ssd1306.setMeasureTimerMode(TIME_INTERVAL)` and  `ssd1306.setMeasureTimerValue(5)` this should not be overwritten. Just want to be sure the logic is working for now, definitely the user does not have to guess all of this by himself :-)

user2684, 2018-11-17:
I've added the following to the Display constructor in #438 so to have at least a default refresh timer set and help users understanding how to customize it:
```
		// refresh the display every minute by default
		setMeasureTimerValue(TIME_INTERVAL);
		setMeasureTimerValue(60UL);
```
I have to keep the report timer to DO_NOT_REPORT otherwise at every refresh a useless message is sent through the radio

closed 2018-11-17 by user2684