mysensors/NodeManager #433: Child::sendValue resets all values, so other sensors cannot access them any more

kainhofer, 2018-10-16:
In the latest development version, the Child::sendValue function calls reset() on the child, which in turn clears all measurements immediately after sending them to the controller.

This conceptual change has the problem that when other sensors (or more important, Displays) are called with their onLoop() / loop() functions, the sensor values of all other sensors are no longer available and e.g. cannot be displayed.

As a quick hack (or rather check whether this is really the problem), I commented out the setReportTimerMode(DO_NOT_REPORT); call in Display::Display and the reset(); call in Child::sendValue. This made the OLED display properly update and show the sensor values of the other sensors on the node...

Of course, that's not the proper solution to the problem, but it proves that this issue and issue #432 together are the reason that OLED displays no longer work.

user2684, 2018-10-21:
This is for sure a serious issue thanks and tells me a need to run a few more tests with displays...
Now, assuming reset has to clear previously stored value (to distinguish between taking measures and reporting), we need to find a way to make this working consistently with any setup. I remember I wanted to keep track of the last value and use that one but I need to remember why I went that way (probably for memory concerns). Let me think about it for a while  :-)

Akubi, 2018-11-01:
@user2684, my feeling is that this was usefull in old version of nodemanager.
Actual version use `_samples `to check if we have data to send.
So probably no more needed to clear `value `in reset function.

@kainhofer, an alternative way, if you use conditional report, is to use `_last_value `instead. But I agree with you, clearing the actual value in reset is probaly not a good idea.

Akubi, 2018-11-03:
@user2684, Ok agree with your explanation. 

We also have the need to report after a _force_update_timer (in case of conditional report). Maby this can also be interesting in non conditional use for refreshing data or other. But il that case we need last value.
 
In any case, I'm not sure to understand the proposal. Can you please gove some more details?

user2684, 2018-11-03:
True, we need to take into account also that scenario but in that case we already have a _last_value variable to use. Back to my proposal, it is something like the following:

```
void Child::_setValueNumber(double value) {
	if (isnan(value)) return;
	if (_samples == 0) {_value = 0; _total = 0;}
	// keep track of the samples
	_samples++;
	...
```
and removing from reset() _value = 0; _total = 0;
In this way _value will stay there for Display and hook functions but will not mess up the calculation for the new measures.

Akubi, 2018-11-03:
@user2684, Sound good.

user2684, 2020-05-04:
Fixed in  #511

closed 2020-05-04 by user2684