mysensors/NodeManager #435: NRF51 implementation has millis() wrap around after ~38 hours, timer no longer triggers

kainhofer, 2018-11-02:
The NRF51 core implementation has a millis() counter that wraps around after ~38 hours. 

Unfortunately, after the wrap-around the main loop is still called, but each sensor's timer will never expire, because in Timer::isOver()
    		if (millis() > _target) return true;
will never match. The _target will be set to a value larger than the wrap-around.

The Timer::isOver function needs to add a detection for millis() wrap around for implementations that do not use the full value range of unsigned long...

Akubi, 2018-11-03:
@kainhofer,

Me again, I yust check the implementation of mills() in the sandeepmistry implementation from 
`https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json`

The radio counter is a 24 buts counter at 32.768kHz but using the overflow int they extend to a full uint32_t value.

```
static volatile uint32_t overflows = 0;

uint32_t millis( void )
{
  uint64_t ticks = (uint64_t)((uint64_t)overflows << (uint64_t)24) | (uint64_t)(NRF_RTC1->COUNTER);
 
  return (ticks * 1000) / 32768;
}
```

Can you please help me understand where do you get the 38 hours ?

user2684, 2018-11-03:
millis() in going to start over sooner or later in every platform so this is something to address I think. In the old version, by keeping track of the old value, this was easy since if millis was less than the previous value, meant it started over. Could be a bit more tricky now and could force us to keep tack of the latest millis somehow which is not ideal but if there are no other choices I'll go for it.

Akubi, 2018-11-03:
@user2684, sure millis() will roll over. But the idea is that if millis() rollover the full range of a uint32_t this is not realy an issue (in my point of view). millis() and _target roll over the same way so a problem can apear after 8 year of continuos work.  Now we can somehow check  the rollover during target update and set/reset a flag in order to manage the special case.

user2684, 2018-11-03:
@Akubi I remember for e.g. an arduino board was something like 50 days (https://www.arduino.cc/reference/en/language/functions/time/millis/), am I wrong? Because if we are talking about 8 years of course I'm not worried at all, if days the thing is different. I wonder if we can do what you are saying above directly into the NodeManager class since update() is called from there we can minimize the impact and keep track of this scenario only once instead of doing it in each timer. How does it sound?

Akubi, 2018-11-03:
@user2684, you are right. 
Arduino is 124 days. 1/3 of year. It's only 62 days if you use long instead of unsigned long.
8 year was from a different implementation (my error).

Concerning the implementation, the problem is that first target will rollover depending on the actual value of millis() and the timer configured _value. So each timer can rollover at a different moment.
When one target value will rollover for a timer, the `if (millis() > _target) return true;` is no more valid.

Maybe a different aproach can be used. If we compute `_target - mills()`, at the begging the result is equal to the Timer configured _value. This value will decrease to zero and then jump to (2^^32) and start decreasing again.
In case of millis() or _target rollover, the result of this operation continue to have the same behavious beacuase the result of operation will also rollover.
So we can now test if the result is > Timer configured _value to detect the Time is over situation.

```
Timer::isOver()
		// check if the current millis() is greater than the target
		if ( (_target - millis()) > _value ) return true;
		return false;
```

Do you agree with this analyses ?

user2684, 2018-11-11:
@akubi I think you are absolutely correct, millis() cannot be smaller than target, this approach should work and makes thing simpler than keeping track of the previous millis in an additional variable. Good idea! 
Changed the label from enhancement to bug.

user2684, 2018-11-17:
@akubi I've implemented it in #438 in a slightly different way:
		// check for a millis() rollover. The difference between millis() and the target (in seconds) cannot be less than value
```
		if ( (_target - millis()) > _value ) return true;
		// check if the current millis() is greater than the target
		if (millis() > _target) return true;
		return false;
```
This is because if only using `if ( (_target - millis()) > _value )`, when timer is over and millis did not roll over, the difference would be negative and the timer never ends. Any test would be appreciated of course :) I'll not merge that PR until I will be sure all the fixes work. Thanks!

Akubi, 2018-11-18:
@user2684, sorry for the late answer, I was in a business trip.
I don't really agree with your sentence : “when timer is over and millis did not roll over, the difference would be negative”
_target and millis() are uint32_t (unsigned) so when  millis()) is bigger than _target , (_target - millis()) will give a value near to 2^32. The test if ( (_target - millis()) > _value ) remain true.
Your second test is dangerous. If you start a timer when millis() + _value generate a rollover, millis() will be > than _target (rolledover) and the timer expire immediately.

user2684, 2018-12-02:
@Akubi sorry, you are fully right, forgot it was unsigned! Thanks for fixing it!