mysensors/NodeManager #436: Homa assistant connection

rspaargaren, 2018-11-05:
would it be possible to create a more easy connection with home assistant. The Hass requires that the sketch presents the values during startup in the loop.
For example for a relay:
  if (!initialValueSent) {
    Serial.println("Sending initial value");
    send(msg.set(state?RELAY_ON:RELAY_OFF));
    Serial.println("Requesting initial value from controller");
    request(CHILD_ID, V_STATUS);
    wait(2000, C_SET, V_STATUS);
  }

user2684, 2018-11-11:
Good idea and I guess other controllers are expecting something similar. Adding it to the dev queue. This will not make it for the new upcoming version since almost ready and only bug fixes are accepted right now but for sure will be part of the future releases. Thanks

RDG88, 2019-03-02:
Would like this one also!

szurr, 2020-05-04:
hi i am trying to connect serial gateway and i only success with relay. should i register sensor in before? or something totaly diffrent

user2684, 2020-05-06:
Hi @szurr is this related to this specific request or is it an unrelated general question? Thanks!

szurr, 2020-05-06:
Above all thank you for this project.
 I guess that my problem with connecting sensors is related to presentation that Home assistant required. But guestion is Rather general.

user2684, 2020-05-10:
This should be fixed by #517 (see PR for implementation details) - both unit prefix and request initial value to the controller. Just keep in mind this is not the default so you need to add a few lines to your code. I'll keep this open until someone would confirm the implementation is ok. Thanks!

szurr, 2020-05-10:
I'll try to check it. After I reinstall my dev home assistant

szurr, 2020-05-11:
Sorry for my lack of understanding but  is hard coded request_initial_value parameter to true is enough? Or should I also set request in sensor class?(I am not quite sure where is it)😁

user2684, 2020-05-11:
For the unit prefix, just call `nodeManager.setSendUnitPrefix(true)` (you can customize the prefix by calling `setUnitPrefix()` to the child).

For requesting initial value, you have to call `setRequestInitialValue(true)` to each child you want to request the initial value from the controller (since this makes sense only for a subset of sensors)

szurr, 2020-05-11:
And I call for each child in presentation function?  I understand this that home assistant need presentation of every sensor at boot and then only changes of state. So I figured that this is only needed at beginning. Are there somewhere any examples where to put those calls?

user2684, 2020-05-12:
Correct, call it on every child but do it in before() (e.g. where the NodeManager code is usually located, have a look at Template.ino for examples). Keep also in mind this feature does not change what NodeManager does in presentation: every child is always presented to the controller, regardless of these settings. Thanks

szurr, 2020-05-12:
hi
i still get compilation error:
exit status 1
'class Child' has no member named 'setRequestInitialValue'; did you mean 'setSendWithoutValue'?
i use for e.g.
void before() {
relay.children.get()->setRequestInitialValue(true)

user2684, 2020-05-15:
Sorry my bad, forgot to mention, ensure you are using code from PR #517 (e.g. https://github.com/user2684/NodeManager/tree/fix/core) otherwise you cannot use the new functionalities. This is because this PR has not been merged yet. Thanks

user2684, 2020-05-15:
Also ensure to configure any sensor *before* calling nodeManager.before(). As for SensorRelay, yes of course you need to configure it with a different pin number for each sensor. Thanks

user2684, 2020-05-23:
#517 has been merged into development so I'm going to close this issue. Feel free to reopen it again if needed

PR #517 (2020-05-23): Core Multiple Fixes
- Added `setSleepBetweenSendSleepOrWait()` to control when `setSleepBetweenSend()` is configured if the node should only wait (default) or sleep or wait depending on the duration set (fixes #456)
- Fixed Interrupt Debounce not working since `_last_interrupt_millis` was never updated (fixes #465)
- Added support for unit prefix consistently across all the sensors (fixes #492 #436):
  - Child constructor can now accept a `unit_prefix `parameter. Setter and getter are also provided (`setUnitPrefix()` / `getUnitPrefix()`). Users can set custom unit prefix by calling `setUnitPre [... cut]
closed 2020-05-23 by user2684