mysensors/NodeManager #442: 'class Sensor' has no member named 'setTrackLastValue'

asoltesz, 2018-12-02:
If I activate FEATURE_CONDITIONAL_REPORT in my sketch (ON), I get a compilation error:

...NodeManagerLibrary.ino: In member function 'virtual void SensorConfiguration::onReceive(MyMessage*)':
NodeManagerLibrary:4192:25: error: 'class Sensor' has no member named 'setTrackLastValue'
         case 7: sensor->setTrackLastValue(request.getValueInt()); break;
                         ^
NodeManagerLibrary:4193:25: error: 'class Sensor' has no member named 'setForceUpdateMinutes'
         case 9: sensor->setForceUpdateMinutes(request.getValueInt()); break;
                         ^
exit status 1
'class Sensor' has no member named 'setTrackLastValue'

Seems like something that remained unchanged from the last refactoring.

user2684, 2018-12-02:
Hi, this should be fixed with the new v1.8 which has just been released. Let me know if it is not the case. Thanks!

asoltesz, 2018-12-10:
The sketch now compiles properly with v1.8. Thanks.

closed 2018-12-10 by asoltesz