mysensors/NodeManager #444: When using SensorBattery Multiple Childs are created

RDG88, 2018-12-04:
I tested the new 1.8 release.
When i enable SensorBattery with the following config:
#include <sensors/SensorBattery.h>
SensorBattery battery(99);

2 childs are created.

- child#3: Type - Battery
- child#99: Type - Battery

Is this by design? in my eyes it should be only child 99

user2684, 2018-12-04:
Hi, no, this shouldn't be case, I think you hit a bug. Can you please share the logs? Thanks!

RDG88, 2018-12-04:
Included my config and log file:

`
 NM:INIT:VER=1.8
0 NM:INIT:INO=PROT-DS18B20-TEST v1.8
0 NM:INIT:LIB VER=2.3.0 CP=RNNNA--- 
0 NM:INIT:RBT p=255
20 NM:BFR:INIT
4676 NM:BFR:OK
4720 NM:PRES:BATTERY(99) p=30 t=38
4758 NM:PRES:BATTERY(1) p=30 t=38
4803 NM:PRES:BATTERY(3) p=30 t=38
6973 NM:STP:ID=222 M=1
7068 NM:STP:HW V=4913 F=17 M=1246
7174 NM:LOOP:BATTERY(99):SET t=38 v=4.91
7296 NM:LOOP:BATTERY(1):SET t=38 v=4.91
7413 NM:LOOP:BATTERY(3):SET t=38 v=4.91
7465 NM:SLP:SLEEP s=10

9591 NM:SLP:AWAKE
9680 NM:LOOP:BATTERY(99):SET t=38 v=4.91
9732 NM:SLP:SLEEP s=10

11858 NM:SLP:AWAKE
11949 NM:LOOP:BATTERY(99):SET t=38 v=4.91
11999 NM:SLP:SLEEP s=10
`

`
#define SKETCH_NAME "PROT-DS18B20-TEST"
#define SKETCH_VERSION "1.8"
#define MY_NODE_ID 222
#define MY_RADIO_NRF24
#define MY_BAUD_RATE 9600
#define MY_SPLASH_SCREEN_DISABLED
#define NODEMANAGER_DEBUG ON
#define NODEMANAGER_INTERRUPTS OFF
#define NODEMANAGER_SLEEP ON
#define NODEMANAGER_RECEIVE OFF
#define NODEMANAGER_DEBUG_VERBOSE OFF
#define NODEMANAGER_POWER_MANAGER OFF
#define NODEMANAGER_CONDITIONAL_REPORT OFF
#define NODEMANAGER_EEPROM OFF
#define NODEMANAGER_TIME OFF
#define NODEMANAGER_RTC OFF
#define NODEMANAGER_SD OFF
#define NODEMANAGER_HOOKING OFF
#define NODEMANAGER_OTA_CONFIGURATION OFF
#define NODEMANAGER_SERIAL_INPUT OFF

// import NodeManager library (a nodeManager object will be then made available)
#include <MySensors_NodeManager.h>

#include <sensors/SensorBattery.h>
SensorBattery battery(99);
SensorBattery setMinVoltage(1.8);
SensorBattery SetMaxVoltage(3.0);

/***********************************
 * Main Sketch
 */

// before
void before() {
	
  /***********************************
   * Configure your sensors
   */
   
  // EXAMPLES:
  // report measures of every attached sensors every 10 seconds
  nodeManager.setReportIntervalSeconds(10);
  // report measures of every attached sensors every 10 minutes
  //nodeManager.setReportIntervalMinutes(5);
  // set the node to sleep in 30 seconds cycles
  nodeManager.setSleepSeconds(10);
  // set the node to sleep in 5 minutes cycles
  //nodeManager.setSleepMinutes(10);
  // report battery level every 10 minutes
  battery.setReportIntervalSeconds(10);
  // set an offset to -1 to a thermistor sensor
  //thermistor.setOffset(-1);
  // change the id of a the first child of a sht21 sensor
  //sht21.children.get(1)->setChildId(5);
  // report only when the analog value is above 40%
  //analog.children.get(1)->setMinThreshold(40);
  // power all the nodes through dedicated pins
//  nodeManager.setPowerManager(power);

  // call NodeManager before routine
  nodeManager.before();
}

// presentation
void presentation() {
  // call NodeManager presentation routine
  nodeManager.presentation();
}

// setup
void setup() {
  // call NodeManager setup routine
  nodeManager.setup();
}

// loop
void loop() {
  // call NodeManager loop routine
  nodeManager.loop();
}

#if NODEMANAGER_RECEIVE == ON
// receive
void receive(const MyMessage &message) {
  // call NodeManager receive routine
  nodeManager.receive(message);
}
#endif

#if NODEMANAGER_TIME == ON
// receiveTime
void receiveTime(unsigned long ts) {
  // call NodeManager receiveTime routine
  nodeManager.receiveTime(ts);
}
#endif
`

user2684, 2018-12-04:
Thanks, I'll try to reproduce it along the weekend and if confirmed I'll evaluate if releasing a hotfix for this since doesn't make sense to wait for next release since v1.8 is just out

RDG88, 2018-12-04:
Thanks for all ur hard work, i love NodeManager

RDG88, 2018-12-07:
Thanks! Working perfectly..