mysensors/NodeManager #456: sleepBetweenSend actually waitBetweenSend?

nikolac, 2018-12-28:
I'm trying to run a few nodes off of super-capacitors and I'm seeing higher than expected voltage drops after transmission. I thought increasing sleepBetweenSend would help, but longer wait was causing bigger drop. 

I noticed that wait() is called instead of sleepOrWait() or sleep(). Is there a reason not to call sleep if the gap is large enough, more than ~500ms?

user2684, 2019-01-06:
Hi, Thanks for reporting this. The behavior of sleepBetweenSend() has been changed with v1.8 because of #405 (ack was lost while sleeping). I'll see if this can be better customized in the next release. Thanks!

user2684, 2020-05-10:
Fixed by #517. Added` setSleepBetweenSendSleepOrWait()` to customize the behavior

PR #517 (2020-05-23): Core Multiple Fixes
- Added `setSleepBetweenSendSleepOrWait()` to control when `setSleepBetweenSend()` is configured if the node should only wait (default) or sleep or wait depending on the duration set (fixes #456)
- Fixed Interrupt Debounce not working since `_last_interrupt_millis` was never updated (fixes #465)
- Added support for unit prefix consistently across all the sensors (fixes #492 #436):
  - Child constructor can now accept a `unit_prefix `parameter. Setter and getter are also provided (`setUnitPrefix()` / `getUnitPrefix()`). Users can set custom unit prefix by calling `setUnitPre [... cut]
closed 2020-05-10 by user2684