mysensors/NodeManager #457: Gateway sending values also sends 0 for own sensors

Hubert2208, 2019-01-03:
I have a ESP8266 gateway with an CCS811 and an SI7021 Sensor. Gateway reports values for both sensors.
If the gateway reports values for a node, at same time it reports 0 values for the own sensors.
For MQTT it looks like this:
mygateway1-out/3/1/1/0/0 -2.30
mygateway1-out/0/1/1/0/0 0.00
mygateway1-out/3/2/1/0/4 971.48
mygateway1-out/0/2/1/0/1 0.00
mygateway1-out/3/3/1/0/5 thunderstorm
mygateway1-out/0/3/1/0/37 0
mygateway1-out/3/4/1/0/0 -2.09
mygateway1-out/0/4/1/0/37 0

i tryed this also with normal gateway-mode. 
Result was exactly the same.

user2684, 2019-01-06:
Thanks for reporting, can you please also share your sketch and NodeManager's logs? Thanks!

Hubert2208, 2019-01-06:
Sure:
here is the Sketch.
It has only a few changes from the original template:
/*
* The MySensors Arduino library handles the wireless radio link and protocol
* between your home built sensors/actuators and HA controller of choice.
* The sensors forms a self healing radio network with optional repeaters. Each
* repeater and gateway builds a routing tables in EEPROM which keeps track of the
* network topology allowing messages to be routed to nodes.
*
* Created by Henrik Ekblad <henrik.ekblad@mysensors.org>
* Copyright (C) 2013-2017 Sensnology AB
* Full contributor list: https://github.com/mysensors/Arduino/graphs/contributors
*
* Documentation: http://www.mysensors.org
* Support Forum: http://forum.mysensors.org
*
* This program is free software; you can redistribute it and/or
* modify it under the terms of the GNU General Public License
* version 2 as published by the Free Software Foundation.
*/

/**************************
Template

This sketch can be used as a template since containing the most relevant MySensors library configuration settings, 
NodeManager's settings, all its the supported sensors commented out and a sketch structure fully functional to operate with
NodeManager. Just uncomment the settings you need and the sensors you want to add and configure the sensors in before()
*/

/**********************************
 * MySensors node configuration
 */

// General settings
#define SKETCH_NAME "MySensorNode"
#define SKETCH_VERSION "1.0"
#define MY_DEBUG
//#define MY_NODE_ID 99

// NRF24 radio settings
//#define MY_RADIO_NRF24
//#define MY_RF24_ENABLE_ENCRYPTION
//#define MY_RF24_CHANNEL 125
//#define MY_RF24_PA_LEVEL RF24_PA_HIGH
//#define MY_DEBUG_VERBOSE_RF24
//#define MY_RF24_DATARATE RF24_250KBPS

// RFM69 radio settings
#define MY_RADIO_RFM69
#define MY_RFM69_FREQUENCY RFM69_868MHZ
#define MY_IS_RFM69HW
#define MY_RFM69_NEW_DRIVER
//#define MY_RFM69_ENABLE_ENCRYPTION
#define MY_RFM69_NETWORKID 100
//#define MY_DEBUG_VERBOSE_RFM69
#define MY_RF69_IRQ_PIN 15
#define MY_RF69_IRQ_NUM MY_RF69_IRQ_PIN
#define MY_RF69_SPI_CS 16
//#define MY_RFM69_ATC_MODE_DISABLED
#define MY_RFM69_MODEM_CONFIGURATION RFM69_FSK_BR4_8_FD9_6

// RFM95 radio settings
//#define MY_RADIO_RFM95
//#define MY_RFM95_FREQUENCY (RFM95_868MHZ)
//#define MY_DEBUG_VERBOSE_RFM95
//#define MY_RFM95_MAX_POWER_LEVEL_DBM (20)
//#define MY_RFM95_IRQ_PIN D1
//#define MY_RFM95_IRQ_NUM MY_RFM95_IRQ_PIN
//#define MY_RFM95_CS_PIN D8
[... 349 lines cut ...]
  // call NodeManager setup routine
  nodeManager.setup();
}

// loop
void loop() {
  // call NodeManager loop routine
  nodeManager.loop();
}

#if NODEMANAGER_RECEIVE == ON
// receive
void receive(const MyMessage &message) {
  // call NodeManager receive routine
  nodeManager.receive(message);
}
#endif

#if NODEMANAGER_TIME == ON
// receiveTime
void receiveTime(unsigned long ts) {
  // call NodeManager receiveTime routine
  nodeManager.receiveTime(ts);
}
#endif

Hubert2208, 2019-01-06:
And the Debug-Log:
24714 NM:INIT:VER=1.8
24737 NM:INIT:INO=MySensorNode v1.0
24775 NM:INIT:LIB VER=2.3.1 CP=RPNGE--- 
24820 MCO:BGN:INIT GW,CP=RPNGE---,REL=255,VER=2.3.1
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 3
cnt 

connected with XXXXxxxx, channel 11
dhcp client start...
ip:192.168.178.20,mask:255.255.255.0,gw:192.168.178.1
25074 MCO:BGN:BFR
25093 NM:INIT:RBT p=255
25118 NM:BFR:INIT
25137 TSF:LRT:OK
25155 TSM:INIT
25171 TSF:WUR:MS=0
25192 TSM:INIT:TSP OK
25215 TSM:INIT:GW MODE
25239 TSM:READY:ID=0,PAR=0,DIS=0
25273 MCO:REG:NOT NEEDED
25301 TSM:READY:NWD REQ
25361 TSF:MSG:SEND,0-0-255-255,s=255,c=3,t=20,pt=0,l=0,sg=0,ft=0,st=OK:
scandone
25801 GWT:TPC:CONNECTING...
25830 GWT:TPC:IP=192.168.178.20
25864 MCO:BGN:STP
25883 NM:STP:ID=0 M=255
26111 NM:STP:HW V=0 F=80 M=45680
26146 MCO:BGN:INIT OK,TSP=1
26175 GWT:TPC:IP=192.168.178.20
26209 GWT:RMQ:MQTT RECONNECT
26256 GWT:RMQ:MQTT CONNECTED
26287 GWT:TPS:TOPIC=mygateway1-out/0/255/0/0/18,MSG SENT
26347 NM:BFR:OK
26364 GWT:TPS:TOPIC=mygateway1-out/0/255/3/0/11,MSG SENT
26425 GWT:TPS:TOPIC=mygateway1-out/0/255/3/0/12,MSG SENT
26485 NM:PRES:SI7021(1) p=6 t=0
26519 GWT:TPS:TOPIC=mygateway1-out/0/1/0/0/6,MSG SENT
26576 NM:PRES:SI7021(2) p=7 t=1
26610 GWT:TPS:TOPIC=mygateway1-out/0/2/0/0/7,MSG SENT
26667 NM:PRES:CCS811(3) p=22 t=37
26703 GWT:TPS:TOPIC=mygateway1-out/0/3/0/0/22,MSG SENT
26762 NM:PRES:CCS811(4) p=22 t=37
26798 GWT:TPS:TOPIC=mygateway1-out/0/4/0/0/22,MSG SENT
26882 NM:LOOP:SI7021(1):SET t=0 v=18.90
26950 NM:LOOP:SI7021(2):SET t=1 v=43.00
26992 GWT:TPS:TOPIC=mygateway1-out/0/1/1/0/0,MSG SENT
27050 GWT:TPS:TOPIC=mygateway1-out/0/2/1/0/1,MSG SENT
27108 NM:SENS:CCS811: ERROR
27137 GWT:IMQ:TOPIC=mygateway1-in/2/255/3/0/6, MSG RECEIVED
28621 !TSF:MSG:SEND,0-0-2-2,s=255,c=3,t=6,pt=0,l=1,sg=0,ft=0,st=NACK:M
28697 NM:LOOP:CCS811(4):SET t=37 v=0
28736 GWT:IMQ:TOPIC=mygateway1-in/0/255/3/0/1, MSG RECEIVED
28799 GWT:IMQ:TOPIC=mygateway1-in/4/255/3/0/6, MSG RECEIVED
30782 !TSF:MSG:SEND,0-0-4-4,s=255,c=3,t=6,pt=0,l=1,sg=0,ft=0,st=NACK:M
30856 GWT:TPS:TOPIC=mygateway1-out/0/4/1/0/37,MSG SENT
30915 GWT:IMQ:TOPIC=mygateway1-in/3/255/3/0/6, MSG RECEIVED
32400 !TSF:MSG:SEND,0-0-3-3,s=255,c=3,t=6,pt=0,l=1,sg=0,ft=0,st=NACK:M
pm open,type:2 0
39197 TSF:MSG:READ,2-2-0,s=255,c=3,t=33,pt=5,l=4,sg=0:300000
39261 GWT:TPS:TOPIC=mygateway1-out/2/255/3/0/33,MSG SENT
40346 TSF:MSG:READ,2-2-0,s=1,c=1,t=0,pt=7,l=5,sg=0:20.09
40406 GWT:TPS:TOPIC=mygateway1-out/2/1/1/0/0,MSG SENT
40464 GWT:TPS:TOPIC=mygateway1-out/0/1/1/0/0,MSG SENT
41469 TSF:MSG:READ,2-2-0,s=2,c=1,t=1,pt=7,l=5,sg=0:42.00
41529 GWT:TPS:TOPIC=mygateway1-out/2/2/1/0/1,MSG SENT
41587 GWT:TPS:TOPIC=mygateway1-out/0/2/1/0/1,MSG SENT
42591 TSF:MSG:READ,2-2-0,s=255,c=3,t=32,pt=5,l=4,sg=0:500
42652 GWT:TPS:TOPIC=mygateway1-out/2/255/3/0/32,MSG SENT
282436 TSF:MSG:READ,3-3-0,s=255,c=3,t=33,pt=5,l=4,sg=0:300000
282501 GWT:TPS:TOPIC=mygateway1-out/3/255/3/0/33,MSG SENT
283633 TSF:MSG:READ,3-3-0,s=1,c=1,t=0,pt=7,l=5,sg=0:1.50
283693 GWT:TPS:TOPIC=mygateway1-out/3/1/1/0/0,MSG SENT
283752 GWT:TPS:TOPIC=mygateway1-out/0/1/1/0/0,MSG SENT
284629 TSF:MSG:READ,3-3-0,s=2,c=1,t=4,pt=7,l=5,sg=0:967.61
284691 GWT:TPS:TOPIC=mygateway1-out/3/2/1/0/4,MSG SENT
284750 GWT:TPS:TOPIC=mygateway1-out/0/2/1/0/1,MSG SENT
285752 TSF:MSG:READ,3-3-0,s=3,c=1,t=5,pt=0,l=6,sg=0:stable
285814 GWT:TPS:TOPIC=mygateway1-out/3/3/1/0/5,MSG SENT
285873 GWT:TPS:TOPIC=mygateway1-out/0/3/1/0/37,MSG SENT
286706 TSF:MSG:READ,3-3-0,s=4,c=1,t=0,pt=7,l=5,sg=0:1.13
286766 GWT:TPS:TOPIC=mygateway1-out/3/4/1/0/0,MSG SENT
286825 GWT:TPS:TOPIC=mygateway1-out/0/4/1/0/37,MSG SENT
287729 TSF:MSG:READ,3-3-0,s=5,c=1,t=1,pt=7,l=5,sg=0:99.00
287790 GWT:TPS:TOPIC=mygateway1-out/3/5/1/0/1,MSG SENT
288849 TSF:MSG:READ,3-3-0,s=255,c=3,t=32,pt=5,l=4,sg=0:500
288911 GWT:TPS:TOPIC=mygateway1-out/3/255/3/0/32,MSG SENT
325934 NM:LOOP:SI7021(1):SET t=0 v=19.13
326003 NM:LOOP:SI7021(2):SET t=1 v=43.00
326046 GWT:TPS:TOPIC=mygateway1-out/0/1/1/0/0,MSG SENT
326105 GWT:TPS:TOPIC=mygateway1-out/0/2/1/0/1,MSG SENT
326166 NM:LOOP:CCS811(3):SET t=37 v=574
326710 NM:LOOP:CCS811(4):SET t=37 v=26
327251 GWT:TPS:TOPIC=mygateway1-out/0/3/1/0/37,MSG SENT
327311 GWT:TPS:TOPIC=mygateway1-out/0/4/1/0/37,MSG SENT
340681 TSF:MSG:READ,4-4-0,s=255,c=3,t=33,pt=5,l=4,sg=0:300000
340746 GWT:TPS:TOPIC=mygateway1-out/4/255/3/0/33,MSG SENT
341829 TSF:MSG:READ,4-4-0,s=1,c=1,t=0,pt=7,l=5,sg=0:15.66
341890 GWT:TPS:TOPIC=mygateway1-out/4/1/1/0/0,MSG SENT
341949 GWT:TPS:TOPIC=mygateway1-out/0/1/1/0/0,MSG SENT
342952 TSF:MSG:READ,4-4-0,s=2,c=1,t=1,pt=7,l=5,sg=0:50.00
343013 GWT:TPS:TOPIC=mygateway1-out/4/2/1/0/1,MSG SENT
343072 GWT:TPS:TOPIC=mygateway1-out/0/2/1/0/1,MSG SENT
344315 TSF:MSG:READ,4-4-0,s=255,c=3,t=32,pt=5,l=4,sg=0:500
344377 GWT:TPS:TOPIC=mygateway1-out/4/255/3/0/32,MSG SENT
366867 TSF:MSG:READ,2-2-0,s=255,c=3,t=33,pt=5,l=4,sg=0:300000
366932 GWT:TPS:TOPIC=mygateway1-out/2/255/3/0/33,MSG SENT
368013 TSF:MSG:READ,2-2-0,s=1,c=1,t=0,pt=7,l=5,sg=0:20.17
368074 GWT:TPS:TOPIC=mygateway1-out/2/1/1/0/0,MSG SENT
368133 GWT:TPS:TOPIC=mygateway1-out/0/1/1/0/0,MSG SENT
369134 TSF:MSG:READ,2-2-0,s=2,c=1,t=1,pt=7,l=5,sg=0:43.00
369195 GWT:TPS:TOPIC=mygateway1-out/2/2/1/0/1,MSG SENT
369254 GWT:TPS:TOPIC=mygateway1-out/0/2/1/0/1,MSG SENT
370255 TSF:MSG:READ,2-2-0,s=255,c=3,t=32,pt=5,l=4,sg=0:500
370317 GWT:TPS:TOPIC=mygateway1-out/2/255/3/0/32,MSG SENT

user2684, 2019-01-09:
Thanks! Sorry just to be sure to fully understand the issue, if the measure is coming from a node is published with that node_id, if coming from a sensor attached to the gateway, it will come from node_id 0. Do you see a different behaviour?

Hubert2208, 2019-01-09:
Hi, if measurements published from sensors attached to the gateway, it will come from node_id 0. Everything is ok.
If measurements published from sensors of any node, they are published with the correct node_id. Additionaly in this case, are published Measurements from same sensor-id of the gateway with payload of 0.

so if measurement from sensors attached to the gateway:
26882 NM:LOOP:SI7021(1):SET t=0 v=18.90
26950 NM:LOOP:SI7021(2):SET t=1 v=43.00
26992 GWT:TPS:TOPIC=mygateway1-out/0/1/1/0/0,MSG SENT
27050 GWT:TPS:TOPIC=mygateway1-out/0/2/1/0/1,MSG SENT

if measurement from a node is published:
41469 TSF:MSG:READ,2-2-0,s=2,c=1,t=1,pt=7,l=5,sg=0:42.00
41529 GWT:TPS:TOPIC=mygateway1-out/2/2/1/0/1,MSG SENT
and additionaly
41587 GWT:TPS:TOPIC=mygateway1-out/0/2/1/0/1,MSG SENT
with payload 0

looks for me like an index-error of a loop.

If you need any further infos -> you are welcome.
Thanks for your great work so far.

Greetings from snowy bavaria
Hubert

user2684, 2019-01-13:
Thanks got it, I'll try to reproduce it and fix it in the next release.

user2684, 2020-05-23:
Hi, I was finally able to look deeper into this and actually the problem is more serious than could have imaged. There is no wrong index, it is more a design issue. 
When a message is received, the NodeManager receive() is called. This is intended mainly for remote sensors so it looks for an attached child to run its loop function. For a gateway, since MySensors does not tell you if a received message is a message addressed to the gateway from a different node or just a sensor reporting (the destination is regardless the gateway), if the child_id is overlapping with one on the gw, its main function is run and then reporting as well. 
I've fixed it through #517 by:

- ignore the message if we are not the final destination (not related to this issue but to add an additional safeguard)
- ignore the message if we are running a gateway and the message is coming from a different node (to avoid triggering our sensors when others are reporting)
Thanks

closed 2020-05-23 by user2684