mysensors/NodeManager #459: collision between declaration for enum timer_mode between Constants.h and libmaple/timer.h

mitchmitchell, 2019-01-20:
timer_mode enum is multpli-defined when building for the STM32F103TB processors -- it might be a good idea to prefect NodeManager types with an "nm_" or something because there are a lot of opportunities for name collisions.

`C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\libraries\NodeManagerLibrary/nodemanager/Constants.h:40:6: error: multiple definition of 'enum timer_mode'

 enum timer_mode {

      ^

In file included from C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple/wirish_types.h:38:0,

                 from C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple/boards.h:38,

                 from C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple/wirish.h:54,

                 from C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\cores\maple/Arduino.h:30,

                 from C:\Myproject\sketch\NodeManager-hytiny-dev.ino.cpp:1:

C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32-master\STM32F1\system/libmaple/include/libmaple/timer.h:549:14: error: previous definition here

 typedef enum timer_mode {

              ^

In file included from C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\libraries\NodeManagerLibrary/nodemanager/Sensor.h:27:0,

                 from C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\libraries\NodeManagerLibrary/nodemanager/Node.h:26,

                 from C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\libraries\NodeManagerLibrary/nodemanager/Node.cpp:24,

                 from C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\libraries\NodeManagerLibrary/MySensors_NodeManager.h:82,

                 from C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\NodeManager-hytiny-dev\NodeManager-hytiny-dev.ino:179:
`

user2684, 2019-01-20:
Hi Mitch, thanks for reporting these three issues, I've marked them as bugs and will be fixed in the next release. This also tells me I'd need to do more tests with different boards before releasing. I hope I can get back to you for testing out the new code once available. Thanks again

user2684, 2020-05-09:
Fixed by #512 renaming the Timer class to InternalTimer to avoid any conflict

closed 2020-05-09 by user2684