mysensors/NodeManager #460: STM32 processor does not have analog reference available, constant DEFAULT not defined

mitchmitchell, 2019-01-20:
In Node.h, the statement uint8_t _analog_reference = DEFAULT; should be protected by an #ifdef CHIP_AVR / #endif pair as the code which uses it is in Node.cpp since the constant DEFAULT is not defined for non AVR processors (or at least not for STM32 processors)

Node.cpp is:
#ifdef CHIP_AVR
	// nothing to do
	if (value == _analog_reference) return;
	// change the analog reference
	analogReference(value);
	// wait a bit 
	wait(200);
	// perform some reading before actually reading the value
	if (pin > -1) {
		for (int i = 0; i < 5; i++) {
			analogRead(pin);
			wait(50);
		}
	}
#endif

Node.h **SHOULD** be as below:
	#ifdef CHIP_AVR
	uint8_t _analog_reference = DEFAULT;
	#endif

Reference regarding ADC accuracy:
https://www.st.com/content/ccc/resource/technical/document/application_note/group0/3f/4c/a4/82/bd/63/4e/92/CD00211314/files/CD00211314.pdf/jcr:content/translations/en.CD00211314.pdf

user2684, 2019-03-10:
For my own reference, related issue and a few other things to fix for this chipset https://forum.mysensors.org/topic/10169/compile-nodemanager-for-stm32f103c8-but-i-get-error/2

user2684, 2020-05-09:
Fixed by #512 (also those fixes from the MySensors forum)

PR #512 (2020-05-09): Multiple fixes for board compatibility
- Moved debug output from NodeManager constructor to before() and removed serial initalization (fix #463 and #503)
- Using MY_WIFI_SSID and MY_WIFI_PASSWORD in template (fix #466)
- Removed CHIP_STM32 interrupt pin definition (fix #461)
- Multiple fixes for STM32 (fix #460)
- Renamed Timer class to InternalTimer (fixes #474)
- Added support for SAMD chips (fix #481)
- Fixed some code not reachable (fix #472)
closed 2020-05-09 by user2684