mysensors/NodeManager #463: On STM32 HyTiny board, NodeManager gets SIGSEGV signal

mitchmitchell, 2019-01-23:
When NodeManager initializes it calls MY_SERIALDEVICE.begin(MY_BAUD_RATE); in its constructor which appears to be before the HardwareSerial object instance has been initialized -- 

This causes usart_init to be called with an argument of dev=0x0 -- which later fails in rb_init

Moving the following code into NodeManager::before() eliminates this problem.   See GDB stack backtrace at the bottom.

```
	#ifndef CHIP_NRF5
		MY_SERIALDEVICE.begin(MY_BAUD_RATE);
	#endif
		// print out the version
		debug(PSTR(LOG_INIT "VER=" VERSION "\n"));
		// print out sketch name
		debug(PSTR(LOG_INIT "INO=" SKETCH_NAME " v" SKETCH_VERSION "\n"));
		// print out MySensors' library capabilities
		debug(PSTR(LOG_INIT "LIB VER=" MYSENSORS_LIBRARY_VERSION " CP=" MY_CAPABILITIES " \n"));

```

Starting program: /home/ubuntu/stm32/HyTiny/NodeManager-hytiny.ino.elf

Program received signal SIGSEGV, Segmentation fault.
rb_init (buf=0x10 <error: Cannot access memory at address 0x10>, size=64, rb=0x8004851 <__start__>)
    at C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/ring_buffer.h:75
75      C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/ring_buffer.h: No such file or directory.
(gdb) backtrace
#0  rb_init (buf=0x10 <error: Cannot access memory at address 0x10>, size=64, rb=0x8004851 <__start__>)
    at C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32\STM32F1\system/libmaple/include/libmaple/ring_buffer.h:75
#1  usart_init (dev=0x0)
    at C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple\libmaple/usart.c:42
#2  0x08004a50 in HardwareSerial::begin (this=0x20000f28 <Serial>, baud=baud@entry=115200, config=config@entry=0 '\000')
    at C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/HardwareSerial.cpp:91
#3  0x08004a8c in HardwareSerial::begin (this=<optimized out>, baud=baud@entry=115200)
    at C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32\STM32F1\cores\maple/HardwareSerial.cpp:69
#4  0x0800181e in NodeManager::NodeManager (this=0x20000d88 <nodeManager>, sensor_count=<optimized out>)
    at C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\libraries\NodeManagerLibrary/nodemanager/Node.cpp:34
#5  0x080039c6 in __static_initialization_and_destruction_0 (__initialize_p=__initialize_p@entry=1,
    __priority=__priority@entry=65535)
    at C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\libraries\NodeManagerLibrary/MySensors_NodeManager.h:85
#6  0x08003ab0 in _GLOBAL__sub_I__Z6hwInitv ()
    at C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\NodeManager-hytiny/NodeManager-hytiny.ino:454
#7  0x08008b5e in __libc_init_array ()
#8  0x08004894 in start_c ()
    at C:\Users\Mitch.MAGNOLIAMANOR\Documents\Arduino\hardware\Arduino_STM32\STM32F1\variants\hytiny_stm32f103t\wirish/start_c.c:84
#9  0xfffffffe in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

user2684, 2019-01-27:
Thanks, as far as I remember that serial begin was before before() because otherwise some of the output generating while instantiating the object would get lost. I wonder if something like https://www.arduino.cc/en/Serial/IfSerial would help with this. Thanks, tagging it as a bug

mitchmitchell, 2019-01-27:
I have been busy building protoboards and getting GDB to resolve windows paths in sources to source files on Linux so I haven't been able to investigate further.  Let me look deeper -- if I remember correctly, I don't think the C++ standards guaranteed any order of invocation of global object constructors that were called before main.  So it may be on STM32s the order of initialization may be different than on Arduinos.  

BTW, is it expected that a user might need to create multiple NodeManager object instances in their program?  If not you might look at making it a static class so only one instance can be created.

mitchmitchell, 2019-01-28:
NodeManager should not need to call MY_SERIALDEVICE.begin because it is called by the Hardware Init routine hwInit on each of the boards I looked at.   I have not yet chased down where hwInit is called relative to the NodeManager global constructor yet.

```
bool hwInit(void)
{
#if !defined(MY_DISABLED_SERIAL)
	MY_SERIALDEVICE.begin(MY_BAUD_RATE);
#if defined(MY_GATEWAY_SERIAL)
	while (!MY_SERIALDEVICE) {}
#endif
#endif
	return true;
}

```

mitchmitchell, 2019-01-28:
Ok, hwInit is calling inside _begin for each of the boards I have looked at, and below you can see that calling the HardwareSerial class before it has constructed itself is going to cause some problems:

The serial device objects are created as globals and if begin is called before these are initialized, there is no this pointer with access to the usart_dev structure.

My next question is why this works on the Arduino -- my guess is that constructors are called in a different order but I don't know for sure yet.  The standard supposedly guarantees that they are called in order of declaration *within* a single translation unit (e.g. object file) once you link multiple object files together the control of the order of invocation is passed to the linker and is not defined in the language standard as far as I know at this point. 

```

HardwareSerial Serial1(USART1, TX1, RX1);

#ifdef TX2
HardwareSerial Serial2(USART2, TX2, RX2);
#endif

#ifdef TX3
HardwareSerial Serial3(USART3, TX3, RX3);
#endif

#if defined(STM32_HIGH_DENSITY) && !defined(BOARD_maple_RET6)
HardwareSerial Serial4(UART4,  TX4, RX4);
HardwareSerial Serial5(UART5,  TX5, RX5);
#endif

HardwareSerial::HardwareSerial(usart_dev *usart_device,
                               uint8 tx_pin,
                               uint8 rx_pin) {
    this->usart_device = usart_device;
    this->tx_pin = tx_pin;
    this->rx_pin = rx_pin;
}

/*
 * Set up/tear down
 */

void HardwareSerial::begin(uint32 baud) {
    ASSERT(baud <= usart_device->max_baud);

    if (baud > usart_device->max_baud) {
        return;
    }

```

user2684, 2019-02-03:
Thanks for the in depth analysis. So basically there is no need for NodeManager to re-initialize the serial port since this is done in _begin() by the library and can cause issues on some boards. Good will go in the direction then.
To your other point, correct, the user is not expected to have multiple NodeManager's instances so a static class is definitely the way to go. Thanks!

leroyle, 2019-10-23:
Just FYI, Probably does not matter but, I can confirm I am also seeing that the NodeManager class is instantiated before the Uart class upon startup, thus I have similar issues as reported initially. Moving the prints to NodeManager::begin() works for this combo as well. 
 My env:
NRF51822,
Platformio     1.39.2
MySensors   2.3.1
NodeManager   1.8.0
gcc version   7.2.1

thanks

user2684, 2020-05-09:
Fixed by #512, thanks all! @mitchmitchell silly question I'm sure but can I ask you how you got the stack trace of the crash initially? (e.g. how to attach gdb, if you have a link to share would be great since would simplify my testing when facing similar problems). Thanks!

closed 2020-05-09 by user2684

mitchmitchell, 2020-05-09:
> Fixed by #512, thanks all! @mitchmitchell silly question I'm sure but can I ask you how you got the stack trace of the crash initially? (e.g. how to attach gdb, if you have a link to share would be great since would simplify my testing when facing similar problems). Thanks!

I used the Black Magic Probe which acts as a GDB client.  It makes connecting with GDB pretty trivial.  https://1bitsquared.com/products/black-magic-probe

user2684, 2020-05-11:
Thanks, interesting, this is new to me!

PR #512 (2020-05-09): Multiple fixes for board compatibility
- Moved debug output from NodeManager constructor to before() and removed serial initalization (fix #463 and #503)
- Using MY_WIFI_SSID and MY_WIFI_PASSWORD in template (fix #466)
- Removed CHIP_STM32 interrupt pin definition (fix #461)
- Multiple fixes for STM32 (fix #460)
- Renamed Timer class to InternalTimer (fixes #474)
- Added support for SAMD chips (fix #481)
- Fixed some code not reachable (fix #472)