mysensors/NodeManager #464: Interrupt pins can be configured as INPUT only without PULLUP/PULLDOWN feature

macs65, 2019-01-24:
The interrupt pins are configured as INPUT in NodeManager::_setupInterrupts() method, this method is called after onSetup of any sensor, than is not possible to change the set of mode of pin to INPUT_PULLUP/INPUT_PULLDOWN in onSetup() sensor method.
The only workarount solution is to change the set of pin in setup() application function after the call of nodeManager.setup()

// setup
void setup() {
	// call NodeManager setup routine
	nodeManager.setup();
        pinMode(INTERRUPT_PIN_1, INPUT_PULLUP); 
}

user2684, 2019-01-27:
Hi, pullup/down can be controlled with `Sensor.setPinInitialValue()` which is intended to take care of it avoiding to set the mode directly to the pins. Is this what you are looking for? Thanks

macs65, 2019-01-27:
Hi,
the function that you have propsed changes initial_value, but the pin mode has been configured in Node.cpp as you cans ee below, the initial value is used _interrupt_1_initial only:
		if (_interrupt_1_mode != MODE_NOT_DEFINED) {
			**pinMode(INTERRUPT_PIN_1,INPUT);**
			if (_interrupt_1_initial > -1) digitalWrite(INTERRUPT_PIN_1,_interrupt_1_initial);

user2684, 2019-02-03:
Sorry for the delay. Apologize, not sure I got it, if _interrupt_1_initial is set e.g. to HIGH through setPinInitialValue() this should be the same as having INPUT_PULLUP (but just for a very short time between the two instructions that you pointed out). Output will then stay HIGH and so you will be able to get notified once it will go LOW but I'm probably missing something. Thanks!