mysensors/NodeManager #465: debounce for interrupt sensors not working

MHofer, 2019-01-24:
setting a higher debounce time has no effect, even if the "setInterruptDebounce" is set to 10000ms, every activity is immediately registered and processed
in the code there is only one assignment for "_last_interrupt_millis" at line 47 of the node.cpp file, in my opinion this variable must also be updated when processing an interrupt otherwise the query at line 525 "(millis () - _last_interrupt_millis <_interrupt_debounce) "always false and the interrupt is processed

did I miss something?

i use version 1.8 and the door sensor

user2684, 2019-01-27:
I think you are right, looks like something got missing in the logic while migrating this capability from v1.7 to v1.8. I'll try to reproduce it and come back in case of doubts, meanwhile I'm tagging this as a bug

user2684, 2020-05-10:
Fixed by #517

PR #517 (2020-05-23): Core Multiple Fixes
- Added `setSleepBetweenSendSleepOrWait()` to control when `setSleepBetweenSend()` is configured if the node should only wait (default) or sleep or wait depending on the duration set (fixes #456)
- Fixed Interrupt Debounce not working since `_last_interrupt_millis` was never updated (fixes #465)
- Added support for unit prefix consistently across all the sensors (fixes #492 #436):
  - Child constructor can now accept a `unit_prefix `parameter. Setter and getter are also provided (`setUnitPrefix()` / `getUnitPrefix()`). Users can set custom unit prefix by calling `setUnitPre [... cut]
closed 2020-05-10 by user2684