mysensors/NodeManager #469: Values not presenting in DisplaySSD1306

user2684, 2019-02-03:
https://forum.mysensors.org/topic/10145/values-not-presenting-in-displayssd1306

greggkn, 2019-04-07:
I have a motion sensor, DHT22, and DisplaySSD1306 connected. I am experiencing the same issue. The values are read from the sensors and sent to the controller. However, the display shows lines for battery, DHT22 temp and humidity, and motion, but they are showing 0 for values.
[image]

user2684, 2019-04-09:
Thanks, can you also share your sketch as well?

ElCheekytico, 2019-04-09:
[TempHumMotionDisplayBatteryNode.txt](https://github.com/mysensors/NodeManager/files/3060218/TempHumMotionDisplayBatteryNode.txt)

ElCheekytico, 2019-04-25:
Any luck with this? Anything I can do to help move this along?

user2684, 2019-04-26:
Hi, development for the new version has not started yet and unfortunately I'm short of spare time to speed this up :-( For sure will put my effort on this issue as the first thing (https://github.com/mysensors/NodeManager/projects/3) but feel free to submit a PR for this if you like. Thanks!

ElCheekytico, 2019-10-31:
Any luck figuring this out?

user2684, 2019-11-01:
Hi, still not on this, sorry. I'm waiting to line up enough issues to start organizing the new development

ElCheekytico, 2020-04-11:
Any progress yet?

user2684, 2020-04-13:
Not yet, really sorry and this has been open for more than a year now so you are completely right in asking for progress...hope this lockdown can help in starting the new development cycle which has still to start. Thanks

user2684, 2020-05-04:
Hi, the root cause of this resulted the same of #433 hence have been just fixed (in the development branch) in #511

closed 2020-05-04 by user2684