mysensors/NodeManager #470: SIGSEGV in dma1() on STM32 HyTiny board

mitchmitchell, 2019-02-08:
I need to investigate in more detail but I have been monitoring my HyTiny board to insure it will run continuiously and it crashes within something like 10 minutes at dma1() -- see backtrace below:

Program received signal SIGSEGV, Segmentation fault.
0x20000084 in dma1 ()
(gdb) backtrace
#0  0x20000084 in dma1 ()
#1  0x08004ae6 in now ()
    at /XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/Arduino/libraries/Time/Time.cpp:262
#2  0x080022d0 in Timer::isOver (this=this@entry=0x20001178)
    at /XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/Arduino/libraries/NodeManagerLibrary/nodemanager/Timer.cpp:90
#3  0x080023c4 in Sensor::_evaluateTimer (
    this=this@entry=0x20000e80 <configuration>, timer=0x20001178)
    at /XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/Arduino/libraries/NodeManagerLibrary/nodemanager/Sensor.cpp:311
#4  0x0800241e in Sensor::loop (this=this@entry=0x20000e80 <configuration>, 
    message=message@entry=0x0)
    at /XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/Arduino/libraries/NodeManagerLibrary/nodemanager/Sensor.cpp:187
#5  0x08003c34 in NodeManager::loop (this=this@entry=0x20000dc4 <nodeManager>)
    at /XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/Arduino/libraries/NodeManagerLibrary/nodemanager/Node.cpp:288
#6  0x08003c60 in loop ()
    at /XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/Arduino/NodeManager-hytiny/NodeManager-hytiny.ino:423
#7  0x08003c7c in main ()
    at /XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX/Arduino/libraries/MySensors/hal/ar--Type <RET> for more, q to quit, c to continue without paging--
chitecture/STM32F1/MyMainSTM32F1.cpp:34
(gdb)

mitchmitchell, 2019-02-08:
Turns out this is not a NodeManager or MySensors issue, its a bug in the SPI library for the STM32 processors -- I've opened an issue on Arduino_STM32 repository.

closed 2019-02-08 by mitchmitchell

user2684, 2019-02-14:
Thanks for closing it down!