mysensors/NodeManager #471: PowerManager is not working for BH1750

RDG88, 2019-03-02:
Hi,

I just tried to use the PowerManager with BH1750. It seems that it does not work.
Did anyone got some luck with this?

Output:
 NM:INIT:VER=1.8
0 NM:INIT:INO=PRO-BH1750-TEST v1.8
0 NM:INIT:LIB VER=2.3.1 CP=RNNNA--- 
0 NM:INIT:RBT p=255
20 NM:BFR:INIT
4401 NM:BFR:OK
4421 NM:PRES:BH1750(1) p=16 t=37
4457 NM:PRES:BATTERY(201) p=30 t=38
6602 NM:STP:ID=5 M=1

user2684, 2019-03-10:
Hi, can you also please post your sketch? Thanks!

RDG88, 2019-04-04:
Sorry for late post.. holidays :-)

`/**********************************
 * MySensors node configuration
 */

// General settings
#define SKETCH_NAME "PRO-BH1750-TEST"
#define SKETCH_VERSION "1.8"
#define MY_NODE_ID 5

//#define MY_DEBUG
// NRF24 radio settings
#define MY_RADIO_NRF24
#define MY_BAUD_RATE 9600
#define MY_SPLASH_SCREEN_DISABLED

/***********************************
 * NodeManager configuration
 */

#define NODEMANAGER_DEBUG ON
#define NODEMANAGER_INTERRUPTS OFF
#define NODEMANAGER_SLEEP ON
#define NODEMANAGER_RECEIVE OFF
#define NODEMANAGER_DEBUG_VERBOSE OFF
#define NODEMANAGER_POWER_MANAGER ON
#define NODEMANAGER_CONDITIONAL_REPORT ON
#define NODEMANAGER_EEPROM OFF
#define NODEMANAGER_TIME OFF
#define NODEMANAGER_RTC OFF
#define NODEMANAGER_SD OFF
#define NODEMANAGER_HOOKING OFF
#define NODEMANAGER_OTA_CONFIGURATION OFF
#define NODEMANAGER_SERIAL_INPUT OFF

// import NodeManager library (a nodeManager object will be then made available)
#include <MySensors_NodeManager.h>

/***********************************
 * Add your sensors
 */

//GND=4 PWR=5
//PowerManager power(4,5);
PowerManager power(4,5);

//#include <sensors/SensorConfiguration.h>
//SensorConfiguration configuration;

//#include <sensors/SensorSignal.h>
//SensorSignal signal;

//#include <sensors/SensorAnalogInput.h>
//SensorAnalogInput analog(A0);

//#include <sensors/SensorLDR.h>
//SensorLDR ldr(A0);

//#include <sensors/SensorRain.h>
//SensorRain rain(A0);

//#include <sensors/SensorSoilMoisture.h>
//SensorSoilMoisture soil(A0);

//#include <sensors/SensorThermistor.h>
//SensorThermistor thermistor(A0);

//#include <sensors/SensorML8511.h>
//SensorML8511 ml8511(A0);
[... 208 lines cut ...]
  // call NodeManager setup routine
  nodeManager.setup();
}

// loop
void loop() {
  // call NodeManager loop routine
  nodeManager.loop();
}

#if NODEMANAGER_RECEIVE == ON
// receive
void receive(const MyMessage &message) {
  // call NodeManager receive routine
  nodeManager.receive(message);
}
#endif

#if NODEMANAGER_TIME == ON
// receiveTime
void receiveTime(unsigned long ts) {
  // call NodeManager receiveTime routine
  nodeManager.receiveTime(ts);
}
#endif`

user2684, 2019-04-09:
thanks, adding to the v1.9 queue for validation

user2684, 2020-05-10:
Hi, the issue seems related to the way you call `setPowerManager()`. For a non-sleeping node like in your examples, calling it to the nodeManager object makes the node to continuously turn power pins on and off since loop is always executed. Try calling `bh1750.setPowerManager(power)` instead which should do the job so pins are turned on/off only upon a measure is retrieved. Feel free to re-open this ticket back again if my understanding is wrong. Thanks!

closed 2020-05-10 by user2684