mysensors/NodeManager #472: Compiler Warnings indicating problems!

g0uus, 2019-03-06:
Hi,
I just downloaded and installed the latest zip version and attempted a build of the AnalogLightAndTemperatureSensor example using Arduino IDE 1.8.8.
I always build with the Compiler Warnings = All setting (as I do in my day job!)
This produces a few warnings that, on investigation are real code issues.

    C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master/nodemanager/Node.cpp: In member function 'void NodeManager::before()':
    C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master/nodemanager/Node.cpp:163:18: warning: comparison is always true due to limited range of data type [-Wtype-limits]
      if (_reboot_pin > -1) {
                      ^
    
    In file included from C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master/MySensors_NodeManager.h:82:0,
                     from C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master\examples\AnalogLightAndTemperatureSensor\AnalogLightAndTemperatureSensor.ino:58:
    C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master/nodemanager/Node.cpp: In member function 'void NodeManager::reboot()':
    C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master/nodemanager/Node.cpp:351:19: warning: comparison is always true due to limited range of data type [-Wtype-limits]
       if (_reboot_pin > -1) {
                       ^
   
    C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master/nodemanager/Node.cpp: In member function 'void NodeManager::_setupInterrupts()':
    C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master/nodemanager/Node.cpp:484:28: warning: comparison is always true due to limited range of data type [-Wtype-limits]
       if (_sleep_interrupt_pin > -1) {
                                ^
   
    C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master/nodemanager/Node.cpp: In member function 'void NodeManager::setAnalogReference(uint8_t, uint8_t)':
    C:\Users\grh\Google Drive\Arduino\libraries\NodeManager-master/nodemanager/Node.cpp:743:10: warning: comparison is always true due to limited range of data type [-Wtype-limits]
      if (pin > -1) {
              ^

In each case these are caused because the `*_pin` variable is declared as `uint8_t` which, by definition, can never be less than zero, hence the code protected by the is statement will ALWAYS be executed.

user2684, 2019-03-10:
Thanks a lot for reporting this! Tagged as a bug and will be fixed in the next release
Thanks again!

user2684, 2020-05-09:
Fixed by #512. This was actually a big issue. Fix was by initializing those variables to 0 (since pin 0 is never valid) so to keep data type

PR #512 (2020-05-09): Multiple fixes for board compatibility
- Moved debug output from NodeManager constructor to before() and removed serial initalization (fix #463 and #503)
- Using MY_WIFI_SSID and MY_WIFI_PASSWORD in template (fix #466)
- Removed CHIP_STM32 interrupt pin definition (fix #461)
- Multiple fixes for STM32 (fix #460)
- Renamed Timer class to InternalTimer (fixes #474)
- Added support for SAMD chips (fix #481)
- Fixed some code not reachable (fix #472)
closed 2020-05-09 by user2684