mysensors/NodeManager #481: 'WDTCSR' was not declared in this scope

Nalith, 2019-05-25:
I keep getting the below error when trying to use NodeManager and a Sensebender Gateway:

------------------------------------
Arduino: 1.8.9 (Windows 10), Board: "Sensebender Gateway"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Nalith\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Nalith\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Nalith\Documents\Arduino\libraries -fqbn=MySensors:samd:mysensors_gw_native -vid-pid=1209_6949 -ide-version=10809 -build-path C:\Users\Nalith\AppData\Local\Temp\arduino_build_722110 -warnings=none -build-cache C:\Users\Nalith\AppData\Local\Temp\arduino_cache_644477 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.openocd-0.9.0-arduino.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.CMSIS.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.0.0-atmel.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.openocd-0.9.0-arduino6-static.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.bossac-1.7.0.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.arduinoOTA-1.2.0.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -prefs=runtime.tools.CMSIS-Atmel-1.1.0.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -verbose C:\Users\Nalith\Source\Repos\Hobby\Home-Automation\MySensors\GatewaySensebender\GatewaySensebender.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Nalith\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Nalith\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Nalith\Documents\Arduino\libraries -fqbn=MySensors:samd:mysensors_gw_native -vid-pid=1209_6949 -ide-version=10809 -build-path C:\Users\Nalith\AppData\Local\Temp\arduino_build_722110 -warnings=none -build-cache C:\Users\Nalith\AppData\Local\Temp\arduino_cache_644477 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.openocd.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.openocd-0.9.0-arduino.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.CMSIS.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.0.0-atmel.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.0.0-atmel -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.bossac.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.openocd-0.9.0-arduino6-static.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino6-static -prefs=runtime.tools.bossac-1.7.0.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.CMSIS-4.5.0.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.arduinoOTA-1.2.0.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.2.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -prefs=runtime.tools.CMSIS-Atmel-1.1.0.path=C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -verbose C:\Users\Nalith\Source\Repos\Hobby\Home-Automation\MySensors\GatewaySensebender\GatewaySensebender.ino
Using board 'mysensors_gw_native' from platform in folder: C:\Users\Nalith\AppData\Local\Arduino15\packages\MySensors\hardware\samd\1.0.5
Using core 'arduino' from platform in folder: C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.21
Detecting libraries used...
"C:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10809 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x1209 -DUSB_PID=0x6949 -DUSBCON "-DUSB_MANUFACTURER=\"MySensors.org\"" "-DUSB_PRODUCT=\"Sensebender GW\"" "-IC:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.1.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.6.21\\cores\\arduino" "-IC:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\MySensors\\hardware\\samd\\1.0.5\\variants\\mysensors_gw" "C:\\Users\\Nalith\\AppData\\Local\\Temp\\arduino_build_722110\\sketch\\GatewaySensebender.ino.cpp" -o nul
"C:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10809 -DARDUINO_SAMD_ZERO -DARDUINO_ARCH_SAMD -D__SAMD21G18A__ -DUSB_VID=0x1209 -DUSB_PID=0x6949 -DUSBCON "-DUSB_MANUFACTURER=\"MySensors.org\"" "-DUSB_PRODUCT=\"Sensebender GW\"" "-IC:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-IC:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\CMSIS-Atmel\\1.1.0/CMSIS/Device/ATMEL/" "-IC:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\samd\\1.6.21\\cores\\arduino" "-IC:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\MySensors\\hardware\\samd\\1.0.5\\variants\\mysensors_gw" "-IC:\\Users\\Nalith\\Documents\\Arduino\\libraries\\NodeManager" "C:\\Users\\Nalith\\AppData\\Local\\Temp\\arduino_build_722110\\sketch\\GatewaySensebender.ino.cpp" -o nul
"C:\\Users\\Nalith\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E  [... cut ...]
Using library Wire at version 1.0 in folder: C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.21\libraries\Wire 
Using library SPI at version 1.0 in folder: C:\Users\Nalith\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.21\libraries\SPI 
Using library Time at version 1.5 in folder: C:\Users\Nalith\Documents\Arduino\libraries\Time 
Using library SD at version 1.2.3 in folder: C:\Program Files (x86)\Arduino\libraries\SD 
exit status 1
Error compiling for board Sensebender Gateway.
------------------------------------

Am I missing something?

user2684, 2019-05-26:
You're right, this is not available on such a board. Tagged as a bug and added to the development queue. Thanks for reporting

Nalith, 2019-05-26:
Great thanks for your prompt reply. Would it work if I just comment out the whole AVR if statement in NodeManager? (Still getting to grips with the code, so this might be a easy answer).

user2684, 2019-05-26:
Absolutely, just comment it out. Thanks

user2684, 2020-05-09:
Fixed by #512 by identifying SAMD chipset

PR #512 (2020-05-09): Multiple fixes for board compatibility
- Moved debug output from NodeManager constructor to before() and removed serial initalization (fix #463 and #503)
- Using MY_WIFI_SSID and MY_WIFI_PASSWORD in template (fix #466)
- Removed CHIP_STM32 interrupt pin definition (fix #461)
- Multiple fixes for STM32 (fix #460)
- Renamed Timer class to InternalTimer (fixes #474)
- Added support for SAMD chips (fix #481)
- Fixed some code not reachable (fix #472)
closed 2020-05-09 by user2684