mysensors/NodeManager #485: PCA9685 not working due to onLoop() is not called

sepo83, 2019-06-10:
The libraries SensorPca9685W, SensorPca9685RGB and SensorPca9685RGBW require their onLoop() function to be called every loop in order to do the color/pwm change. It seems that in V1.8 of NodeManager this is not done every loop.

Workaround: Call the onLoop()-function directely in the loop-funtion of the sketch. For example:
`led_b1_rgb.onLoop(NULL);`

Is there any "cleaner" solution?

user2684, 2019-06-12:
Hi, thanks for reporting this issue. Theoretically onLoop() is called at any loop() since the latter is just wrapping the first one unless skipped because sleeping or because of a reporting timer set. Isn't it the case? Thanks

sepo83, 2019-06-12:
After doing some debugging I think it has something to do with the reporting timer. Every loop I get `_evaluateTimer(_measure_timer)=0, _measure_timer=1174, message=0`
(debug print was set at the first line of Sensor::loop)

Do I have to set this timer somehow?

I defined the following:
`#define NODEMANAGER_DEBUG ON`
`#define NODEMANAGER_INTERRUPTS ON`
`#define NODEMANAGER_SLEEP OFF`
`#define NODEMANAGER_RECEIVE ON`
`#define NODEMANAGER_DEBUG_VERBOSE ON`
`#define NODEMANAGER_POWER_MANAGER OFF`
`#define NODEMANAGER_CONDITIONAL_REPORT OFF`
`#define NODEMANAGER_EEPROM OFF`
`#define NODEMANAGER_TIME OFF`
`#define NODEMANAGER_RTC OFF`
`#define NODEMANAGER_SD OFF`
`#define NODEMANAGER_HOOKING OFF`
`#define NODEMANAGER_OTA_CONFIGURATION OFF`
`#define NODEMANAGER_SERIAL_INPUT OFF`

I configured/set nothing else in before().

user2684, 2019-07-22:
Theoretically, when you set a report timer and if a measure timer is not set explicit, it is just copied over. However, you can set just the measure timer if you wish with:
```
Sensor::setMeasureTimerMode(timer_mode value)
Sensor::setMeasureTimerValue(unsigned long value)
```
Something like:
```
setMeasureTimerMode(TIME_INTERVAL)
setMeasureTimerValue(60)
```
This is because there is no equivalent for setReportIntervalSeconds() for measure timer, you have to call both explicitly. Try give it a try, if it works, we will add this into this sensor code so the user does not have to make it explicit. Thx