mysensors/NodeManager #488: MPR121 won't send value to controller

Bottesford, 2019-07-20:
I'm having trouble with the touch sensor sensor you kindly implemented on my request in 1.8. I can see my touch inputs are being recorded ok in the serial output but no matter what I try I can't get the sketch to transmit the values to the gateway. In fact if I combine the MPR121 sensor with a motion sensor it too won't send values back either (but will work fine on its own).
Using the standard template sketch the only important bits are..
```
#include <sensors/SensorMPR121.h>
SensorMPR121 mpr121(2);

void before() {
  // setup the serial port baud rate
  Serial.begin(MY_BAUD_RATE);
  
  mpr121.setPasscodeLength(1);  // One press, one send to controller

   // call NodeManager before routine
  nodeManager.before();
}
```

The serial log for me pressing various buttons over a few seconds:
146120 NM:SENS:MPR121(2):READ v=11
146157 NM:LOOP:MPR121(2):SET t=47 v=11
146198 NM:SENS:MPR121(2):READ v=11
151236 NM:SENS:MPR121(2):READ v=9
151272 NM:LOOP:MPR121(2):SET t=47 v=9
151312 NM:SENS:MPR121(2):READ v=9
152648 NM:SENS:MPR121(2):READ v=6
152684 NM:LOOP:MPR121(2):SET t=47 v=6
152724 NM:SENS:MPR121(2):READ v=6
153460 NM:SENS:MPR121(2):READ v=4
153496 NM:LOOP:MPR121(2):SET t=47 v=4
153536 NM:SENS:MPR121(2):READ v=4

So the value is set to the child ok but no matter how long we wait or keep trying nothing ever goes to the gateway. I've fiddled with the passcode length (here set to 4) and found it behaves as it should:

60964 NM:SENS:MPR121(2):READ v=11
63800 NM:SENS:MPR121(2):READ v=5
64535 NM:SENS:MPR121(2):READ v=6
64570 NM:SENS:MPR121(2):READ v=7
64605 NM:LOOP:MPR121(2):SET t=47 v=11567
64647 NM:SENS:MPR121(2):READ v=7
66882 NM:SENS:MPR121(2):READ v=3
66917 NM:SENS:MPR121(2):READ v=4
66952 NM:SENS:MPR121(2):READ v=5
67386 NM:SENS:MPR121(2):READ v=6
67421 NM:LOOP:MPR121(2):SET t=47 v=3456
67463 NM:SENS:MPR121(2):READ v=6
67996 NM:SENS:MPR121(2):READ v=5
68031 NM:SENS:MPR121(2):READ v=6
68267 NM:SENS:MPR121(2):READ v=5
69001 NM:SENS:MPR121(2):READ v=4
69036 NM:LOOP:MPR121(2):SET t=47 v=5654
69078 NM:SENS:MPR121(2):READ v=4
69911 NM:SENS:MPR121(2):READ v=6
71946 NM:SENS:MPR121(2):READ v=11
73282 NM:SENS:MPR121(2):READ v=11
74318 NM:SENS:MPR121(2):READ v=11
74354 NM:LOOP:MPR121(2):SET t=47 v=7221

Except it never gets transmitted even after 4 buttons pressed.

And now with the motion sensor:
48228 NM:LOOP:INT p=2 v=1
52306 NM:LOOP:INT p=2 v=0
57444 NM:LOOP:INT p=2 v=1
57444 NM:LOOP:INT p=2 v=1
62021 NM:LOOP:INT p=2 v=0
72848 NM:SENS:MPR121(2):READ v=11
72884 NM:LOOP:MPR121(2):SET t=47 v=11
72924 NM:SENS:MPR121(2):READ v=11
73959 NM:SENS:MPR121(2):READ v=11
73995 NM:LOOP:MPR121(2):SET t=47 v=11
74035 NM:SENS:MPR121(2):READ v=11
74470 NM:SENS:MPR121(2):READ v=11
74506 NM:LOOP:MPR121(2):SET t=47 v=11
74546 NM:SENS:MPR121(2):READ v=11
79065 NM:LOOP:INT p=2 v=1
81459 NM:LOOP:INT p=2 v=0
88551 NM:LOOP:INT p=2 v=1
91644 NM:LOOP:INT p=2 v=0

Sensors both working fine but no transmit. I use this setting for my pir:
`motion.setInterruptMode(CHANGE);`

I've tried this (but I know this is set in the sensor already):
`mpr121.setReportTimerMode(IMMEDIATELY);`

I've also tried setting the report period like this with no joy:
`nodeManager.setReportIntervalSeconds(2);`

All nodemanager features are off except debug (+ interrupt when using motion sensor). However I've tried many combos and fiddled with conditional reporting settings and I just can't get it to go.
Other sensors (motion, sonar, door/windows, waterflow, temperature, humidity, light level..) all work properly on their own or in combination, so it's just the mpr121 for some reason.

My workaround has been to add a sendValue call on the child within the SensorMPR121 onLoop method:
`child->setValue(passcode);
child->sendValue(true);`

Clearly not how you have intended this to work but gets it going.
Unfortunately adding a sendValue in SensorInterrupt doesn't fix it for the motion sensor so think i'll have to stick to my custom mpr121 implementation for now.

user2684, 2019-07-22:
Hi, weird also because especially the motion sensor has been tested many many times and always working. The logs tell us everything is apparently working as expected with values read and finally something set to the child (which is https://github.com/mysensors/NodeManager/blob/master/sensors/SensorMPR121.h#L103).
Looks like for some reason sendValue() is not called even with setReportTimerMode(IMMEDIATELY) which should not wait for additional condition. 
I need to run a few tests to reproduce the problem, meanwhile let me tag it as a bug and add to v1.9 dev queue. Thanks

user2684, 2020-05-09:
Fixed by #514

closed 2020-05-09 by user2684