mysensors/NodeManager #489: DHT11 not sending any humidity data

pleschev, 2019-08-06:
Setting up a basic NodeManager node with just a SensorDHT11 defined, humidity data isn't being sent from the sensor:

```
#include "sensors/SensorDHT11.h"
SensorDHT11 sensorDHT11(3);
```
Digging around the Nodemanager code:
https://github.com/mysensors/NodeManager/blob/master/sensors/SensorDHT.h#L54
I noticed that on the first call to onLoop() for the temperature, it all works correctly, on the second call of onLoop() for humidity, the sensor library returns NaN. I got it working by increasing the wait time from:
`nodeManager.sleepOrWait(_dht->getMinimumSamplingPeriod());
`
to:
`
nodeManager.sleepOrWait(_dht->getMinimumSamplingPeriod()+100);=
`
This fixes the problem for me, however, I don't know what the best fix would be that is worthy of a pull request

user2684, 2019-08-09:
Thanks for reporting this, tagging it as a bug. Yeah, DHT sensors are always tricky to pull from, adding this additional waiting time is definitely safer

user2684, 2020-05-04:
Fixed with #510

closed 2020-05-04 by user2684