mysensors/NodeManager #492: best way to set units via V_UNIT_PREFIX ?

zerog2k, 2019-10-29:
Hello,
I have nodemanager sketch using SensorSoilMoisture with Home Assistant, and would like to "hint" that the units are in "%", such that HA displays the history graphs correctly (timeseries rather than some strange bar-graph series). From mysensors/ha docs, seems that we do this by sending the units as some V_UNIT_PREFIX.

So looking for the suggestion on best pattern for sending V_UNIT_PREFIX, per 
https://www.home-assistant.io/integrations/sensor.mysensors/#mysensors-2x-example-sketch

I think that I just need to get the node's child id, and then send() that at least once, but not exactly sure how to do this. Perhaps this should actually be in SensorSoilMoisture ? (i.e. send units as % ?)

zerog2k, 2019-10-29:
PR #342 touches on this, but never provides a resolution to it

user2684, 2019-11-01:
You have a point, so far V_UNIT_PREFIX has not been implemented consistently across the sensors. I'd figure out a way to give the ability to set the unit prefix for each child so that NodeManager will take care of it. Will probably take a while since not necessarily an easy one. Unfortunately I do not have a decent workaround. But I'm wondering if you can add a custom hook which aritifically sends the prefix

user2684, 2020-05-10:
Fixed by #517. See PR for implementation details.

PR #517 (2020-05-23): Core Multiple Fixes
- Added `setSleepBetweenSendSleepOrWait()` to control when `setSleepBetweenSend()` is configured if the node should only wait (default) or sleep or wait depending on the duration set (fixes #456)
- Fixed Interrupt Debounce not working since `_last_interrupt_millis` was never updated (fixes #465)
- Added support for unit prefix consistently across all the sensors (fixes #492 #436):
  - Child constructor can now accept a `unit_prefix `parameter. Setter and getter are also provided (`setUnitPrefix()` / `getUnitPrefix()`). Users can set custom unit prefix by calling `setUnitPre [... cut]
closed 2020-05-10 by user2684