mysensors/NodeManager #496: SensorBattery measure internal VCC on ATmega1284

j54n1n, 2020-01-04:
Hello there,

I am currently on the latest release 1.8 of NodeManager and I am trying to use it on a board with ATmega1284p chip on it.

I include the SensorBattery to measure the VCC voltage as suggested in the template sketch:
```
#include <sensors/SensorBattery.h>
SensorBattery battery;
```
but I get an error message:
```
C:\Files\Work\arduino\libraries\NodeManager/sensors/SensorBattery.h:91:35: error: 'INTERNAL' was not declared in this scope
    nodeManager.setAnalogReference(INTERNAL);
                                   ^~~~~~~~
```
In fact if I look in the [Arduino.h](https://github.com/arduino/ArduinoCore-avr/blob/366b1b814702ec0da8951f9e13af882352442940/cores/arduino/Arduino.h#L77..L82) file there INTERNAL gets replaced by INTERNAL1V1:
```
#if defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__) || defined(__AVR_ATmega1284__) || defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644__) || defined(__AVR_ATmega644A__) || defined(__AVR_ATmega644P__) || defined(__AVR_ATmega644PA__)
#define INTERNAL1V1 2
#define INTERNAL2V56 3
#else
#define INTERNAL 3
#endif
```
And if I look at the [SensorBattery.h](https://github.com/mysensors/NodeManager/blob/f5204083628d72175a89033af7bfc222cf424172/sensors/SensorBattery.h#L90..L95) file I can see that there is the possibility to switch to INTERNAL1V1:
```
#ifdef CHIP_AVR
			nodeManager.setAnalogReference(INTERNAL);
#endif
#ifdef CHIP_MEGA
			nodeManager.setAnalogReference(INTERNAL1V1);
#endif
```
Therefore it should be possibile to edit [Constants.h](https://github.com/mysensors/NodeManager/blob/0e3203ac17b629c0c82e2228c815aa1b5f9cd911/nodemanager/Constants.h#L88..L89) file on line 88:
```
#if defined(__AVR_ATmega32U4__) || defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
#define CHIP_MEGA
```
and add the other define checks at the end like so:
```
#if defined(__AVR_ATmega32U4__) || defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__) || defined(__AVR_ATmega1284__) || defined(__AVR_ATmega1284P__) || defined(__AVR_ATmega644__) || defined(__AVR_ATmega644A__) || defined(__AVR_ATmega644P__) || defined(__AVR_ATmega644PA__)
#define CHIP_MEGA
```
And it should just work.
Could you include this fix in the next release?

j54n1n, 2020-01-04:
Ah and I forgot that in the [SensorBattery.h](https://github.com/mysensors/NodeManager/blob/0e3203ac17b629c0c82e2228c815aa1b5f9cd911/sensors/SensorBattery.h#L90..L95) file from line 90:
```
#ifdef CHIP_AVR
			nodeManager.setAnalogReference(INTERNAL);
#endif
#ifdef CHIP_MEGA
			nodeManager.setAnalogReference(INTERNAL1V1);
#endif
```
should be changed to:
```
#if defined(CHIP_MEGA)
			nodeManager.setAnalogReference(INTERNAL1V1);
#elif defined(CHIP_AVR)
			nodeManager.setAnalogReference(INTERNAL);
#endif
```

user2684, 2020-01-05:
Thanks for reporting this! Definitely will be fixed in the next release

j54n1n, 2020-01-06:
> 
> 
> Thanks for reporting this! Definitely will be fixed in the next release

Thanks @user2684 ^^

user2684, 2020-05-04:
Fixed by #509

PR #509 (2020-05-04): Fixed analog reference compilation error with ATmega1284
Fixes #496
closed 2020-05-04 by user2684