mysensors/NodeManager #497: bmp280 as dependency issue.

deisi, 2020-01-04:
I try to use a bmp280 with the development branch, but I run into a dependency issue.
The readme states, that I would need https://github.com/adafruit/Adafruit_BMP280_Library.
However, I have it installed, and it is not enough. Building fails with:

```
/home/malte/Nextcloud/Arduino/libraries/MySensors_NodeManager/sensors/SensorBMP280.h:27:10: fatal error: Adafruit_Sensor.h: No such file or directory
   27 | #include <Adafruit_Sensor.h>
      |          ^~~~~~~~~~~~~~~~~~~
```

If you inspect, the library, there is no `Adafruit_Sensor.h` file. Anybody knows witch header file is needed here and where to get it?

deisi, 2020-01-04:
Okay, found it. https://github.com/adafruit/Adafruit_Sensor/releases should be added in the readme as well I think.

user2684, 2020-01-05:
Hi, yes, most of Adafruit's libraries require Adafruit Sensor but is not written anywhere. Will update the documentation accordingly. Thanks!

user2684, 2020-05-09:
Fixed by #515

closed 2020-05-09 by user2684