mysensors/NodeManager #498: The sensor order of BH1750 and BME280 matters

deisi, 2020-01-07:
I just realized, that wen I use:

```
#include <sensors/SensorBME280.h>
SensorBME280 bme280;

#include <sensors/SensorBH1750.h>
SensorBH1750 bh1750;

```

Both sensors work. However if I switch the order:

```

#include <sensors/SensorBH1750.h>
SensorBH1750 bh1750;

#include <sensors/SensorBME280.h>
SensorBME280 bme280;
```

both sensors fail during there configuration. Is this know or intended?

user2684, 2020-01-29:
Hi, no, it is not intended, which makes me think of some kind of incompatibility. Do you have the compilation error to share? Thanks

deisi, 2020-01-29:
Well its not the compiler that doesn't work, but the node would put out gibberish data or it doesn't run. I don't remember right now what the problem was I think I could make another test node at the weekend and dig a little deeper.

user2684, 2020-02-10:
I can recall a similar situation with BME happening to me as well, when combining sensors I got unusable data, but that situation was due to a board almost running out of memory. If you have both compiler and runtime logs I would be happy to have a deeper look. Thanks

user2684, 2020-05-09:
Hi, I couldn't reproduce the issue unfortunately. My feeling is that is could have something to do with the storage utilization (runtime) which the adafruit BMP/BME library loves to eat up. I'd recommend give a try to the new define NODEMANAGER_SENSOR_BOSCH_LITE introduced in #515 which should reduce significantly the memory used (by excluding the forecast child). Feel free to reopen this issue in case you have any news or comment. Thanks!

closed 2020-05-09 by user2684