mysensors/NodeManager #501: Multiple samples averaging error

nielbierman, 2020-02-03:
I have found that when requesting more than one read to average by using setSamples(), the reading being sent is always the last reading and not the average of the samples.
The value sent in the attached example was 213...the last of the 30 values read. The average is actually 209.6
[image]

user2684, 2020-02-10:
Thanks for reporting this! I tag it as a bug and fix in the next release

user2684, 2020-05-04:
After reviewing the code, this looks like related to the way values are processed for each child when per-child processing policies have been introduced. The default is to keep the latest value only, this is why your sampling is not working. Try calling setValueProcessing(AVG) to the child object and let me know. Thanks!

closed 2020-05-10 by user2684