mysensors/NodeManager #502: SensorMotion not working

RolandWijnen, 2020-04-05:
Hi, 

First of all a big thank you for creating NodeManager! Makes it so much simpler to combine sensors for the different nodes I'm making. 

I'm however a bit stuck with integrating the MotionSensor. I'm using Arduino IDE 1.8.12, MySensors 2.3.0, and NodeManager 1.8. I've included a DHT11, MQ135, and RELAY and these are all working great. The motion sensor does however not work. 

I've tested the sensor by connecting it to another node with only the motion sensor attached. If I upload the default MySensors MotionSensor example, the node works and the sensor responds to movement. If I upload the NodeManager code, it does not work. It seems to be set to a "HIGH" value and not responding to movement. I've fiddled around with the code like this: 

`motion.setPinInitialValue("LOW");`
`motion.setInterruptMode("CHANGE");`
`// call NodeManager before routine`
`nodeManager.before();`

This sets the sensor to LOW as expected, but still, the sensor does not respond. 

I also find it strange that my sensor reports a "HIGH" value in the first place, since the code in SensorMotion.h initializes the pin as "LOW". 

Can anybody shed some light on what I'm missing here? I've been puzzled about this for a couple of days now, and cannot find any clue on how to fix this in the issues (open and closed) reported here.

Thanks, Roland.

user2684, 2020-04-13:
Hi, weird behaviour, wonder if by mistake you are using the same pin with other sensors which are changing the status of the pin. Also, is the pin really high with a multimeter? If you have any log and/or the sketch to share would be great. Thanks

RolandWijnen, 2020-04-20:
Hi user2684, 

I've dpne some checking and testing with a multimeter. It seems that the motion sensor is working after all. It is low by default and becomes high as it should. No bug after all. My misstake. This is the output from the serial terminal:

> 23:48:25.762 -> 0 NM:INIT:INO=MySensors_WoodStorage v1.0
23:48:25.829 -> 0 NM:INIT:LIB VER=2.3.2 CP=RNNNA--- 
23:48:25.867 ->  
23:48:25.867 ->  __  __       ____
23:48:25.902 -> |  \/  |_   _/ ___|  ___ _ __  ___  ___  _ __ ___
23:48:25.939 -> | |\/| | | | \___ \ / _ \ `_ \/ __|/ _ \| `__/ __|
23:48:25.975 -> | |  | | |_| |___| |  __/ | | \__ \  _  | |  \__ \
23:48:26.043 -> |_|  |_|\__, |____/ \___|_| |_|___/\___/|_|  |___/
23:48:26.114 ->         |___/                      2.3.2
23:48:26.152 -> 
23:48:26.152 -> 209 NM:INIT:RBT p=255
23:48:26.152 -> 14680364 NM:BFR:INIT
23:48:30.493 -> 14684640 NM:BFR:OK
23:48:30.526 -> 14684684 NM:PRES:MOTION(1) p=1 t=16
23:48:30.986 -> 14685133 NM:STP:ID=6 M=1
23:48:30.986 -> 14685159 NM:BFR:INT p=3 m=3
23:48:31.661 -> 14685259 NM:STP:HW V=5068 F=17 M=1463
23:48:31.697 -> 14685299 NM:SLP:SLEEP s=3600
23:48:31.730 -> 
23:48:48.274 -> 14685846 NM:LOOP:INT p=3 v=1
23:48:48.308 -> 14685877 NM:SLP:AWAKE
23:48:48.308 -> 14685900 NM:LOOP:MOTION(1):SET t=16 v=1
23:48:48.375 -> 14685943 NM:SLP:SLEEP s=3600
23:48:48.408 -> 
23:49:00.945 -> 14686485 NM:LOOP:INT p=3 v=1
23:49:00.983 -> 14686516 NM:SLP:AWAKE
23:49:00.983 -> 14686539 NM:LOOP:MOTION(1):SET t=16 v=1
23:49:01.018 -> 14686582 NM:SLP:SLEEP s=3600
23:49:01.056 -> 

I'm using the example code for the motionsensor and did not change anything. 

I was misled by looking at Home Assistant output and that keeps the state of the motionsensor to 'Detected' all the time. I must have a setting wrong there I guess. 

Or does the MotionSensor example not report 'no motion' to the gateway? 

Cheers, Roland.

closed 2020-04-20 by RolandWijnen