mysensors/NodeManager #503: Issues compiling for NRF5x

user2684, 2020-04-18:
Ref. https://forum.mysensors.org/topic/10315/nrf5x-on-nodemanager

puneit-hassio, 2020-04-19:
Hi, 
I changed the radio define line for the NRF5x. And normally, we are required to have two tabs containing MyBoardNRF5.cpp and MyBoardNRF5.h along with the .ino file when compiling for this board. Getting the following errors

`Arduino: 1.8.11 (Mac OS X), Board: "MyBoardNRF5 nRF51822, 16 kB RAM, 256 kB flash (xxaa), None, Crystal Oscillator"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/puneit/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/puneit/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/puneit/Documents/Arduino/libraries -fqbn=MySensors:nRF5:MyBoard_nRF51822:chip=xxaa,bootcode=none,lfclk=lfxo -ide-version=10811 -build-path /var/folders/vx/9fdfhc216_x67q1q0vhs7fsm0000gn/T/arduino_build_871442 -warnings=none -build-cache /var/folders/vx/9fdfhc216_x67q1q0vhs7fsm0000gn/T/arduino_cache_719849 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.gcc-arm-none-eabi.path=/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4 -prefs=runtime.tools.gcc-arm-none-eabi-5_2-2015q4.path=/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4 -prefs=runtime.tools.openocd.path=/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5 -prefs=runtime.tools.openocd-0.10.0-dev.nrf5.path=/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5 -verbose /Users/puneit/Google Drive/MySensor/NRF51822/Nodemanager/Template/Template/Template.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/puneit/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/puneit/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/puneit/Documents/Arduino/libraries -fqbn=MySensors:nRF5:MyBoard_nRF51822:chip=xxaa,bootcode=none,lfclk=lfxo -ide-version=10811 -build-path /var/folders/vx/9fdfhc216_x67q1q0vhs7fsm0000gn/T/arduino_build_871442 -warnings=none -build-cache /var/folders/vx/9fdfhc216_x67q1q0vhs7fsm0000gn/T/arduino_cache_719849 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.gcc-arm-none-eabi.path=/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4 -prefs=runtime.tools.gcc-arm-none-eabi-5_2-2015q4.path=/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4 -prefs=runtime.tools.openocd.path=/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5 -prefs=runtime.tools.openocd-0.10.0-dev.nrf5.path=/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/openocd/0.10.0-dev.nrf5 -verbose /Users/puneit/Google Drive/MySensor/NRF51822/Nodemanager/Template/Template/Template.ino
Using board 'MyBoard_nRF51822' from platform in folder: /Users/puneit/Library/Arduino15/packages/MySensors/hardware/nRF5/0.3.0
Using core 'nRF5' from platform in folder: /Users/puneit/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0
Detecting libraries used...
/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10811 -DARDUINO_GENERIC -DARDUINO_ARCH_NRF5 -DNRF51 -DMYBOARDNRF5 -I/var/folders/vx/9fdfhc216_x67q1q0vhs7fsm0000gn/T/arduino_build_871442 -DNRF5 -I/Users/puneit/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/toolchain/ -I/Users/puneit/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/toolchain/CMSIS/Include -I/Users/puneit/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/toolchain/gcc/ -I/Users/puneit/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/device/ -I/Users/puneit/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/drivers_nrf/delay/ -I/Users/puneit/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5/SDK/components/softdevice/none/headers/ -DUSE_LFXO -I/Users/puneit/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/cores/nRF5 -I/Users/puneit/Library/Arduino15/packages/MySensors/hardware/nRF5/0.3.0/variants/MyBoardNRF5 /var/folders/vx/9fdfhc216_x67q1q0vhs7fsm0000gn/T/arduino_build_871442/sketch/Template.ino.cpp -o /dev/null
Alternatives for MySensors_NodeManager.h: [NodeManager-master@1.8.0]
ResolveLibrary(MySensors_NodeManager.h)
  -> candidates: [NodeManager-master@1.8.0]
/Users/puneit/Library/Arduino15/packages/sandeepmistry/tools/gcc-arm-none-eabi/5_2-2015q4/bin/arm-none-eabi-g++ -mcpu=cortex-m0 -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -DF_CPU=16000000 -DARDUINO=10811 -DARDUINO_GENERIC -D [... cut ...]
Using library NodeManager-master at version 1.8.0 in folder: /Users/puneit/Documents/Arduino/libraries/NodeManager-master 
Using library MySensors-master at version 2.3.2 in folder: /Users/puneit/Documents/Arduino/libraries/MySensors-master 
Using library SPI at version 1.0 in folder: /Users/puneit/Library/Arduino15/packages/sandeepmistry/hardware/nRF5/0.6.0/libraries/SPI 
exit status 1
Error compiling for board MyBoardNRF5 nRF51822.`
[Template.zip](https://github.com/mysensors/NodeManager/files/4498455/Template.zip)

user2684, 2020-04-25:
Thanks, I was finally able to reproduce the problem. Looks like the issue is related to the newest version of the MySensors library. Give a try with 2.3.1 instead of 2.3.2, to me compiled just fine. If confirmed of course I urgently need to support 2.3.2 as well since looks like something has changed under the hood making it not fully compatible with NodeManager. Thanks!

puneit-hassio, 2020-04-27:
Hi, Yes, it compiles with 2.3.1...but there is no output on the serial monitor and nothing on the MQTT traffic as well. 
[nodemanagertest_lib2.3.1.zip](https://github.com/mysensors/NodeManager/files/4539029/nodemanagertest_lib2.3.1.zip)

user2684, 2020-05-03:
Ok, at least this is a step ahead, unfortunately fixing compilation issues is way easier than reproducing runtime issues :-) Let me try setting up a test board and see if I can get somewhere. Do you have any *working* code to share (just mysensors, without NodeManager) for this board so I can try to find the differences. Thx

user2684, 2020-05-03:
Nevermind, I was eventually able to reproduce it but unfortunately not fixing it. As you may have noticed, the problem is not in the serial output but looks like a runtime error is making the sketch crashing hence no output is returned. I spent long time trying to isolate the problem and the first candidate is NodeManager's debug() which is mapping directly MySensors HwDebugPrint(). The latter is used by MySensors core without issues but when I call it (actually whenever I use a Serial.print), there is apparently a crash. I couldn't find an explanation yet and debugging it could be time consuming. If I could only get some sort of stack trace from the board upon crash....

puneit-hassio, 2020-05-04:
If there is something that you need me to try or do ; let me know

user2684, 2020-05-06:
Just if you have a better way for testing rather than flashing with st-linkv2, disconnecting, connecting  to a serial TTL and run it since it is very time consuming going through this process to verify a single line commented out is making it crashing or not. Is there a way to flash and connect to serial like a standard arduino board without using two different USB devices? Thanks

Unfortunately it is very manual and try-and-error the process with this since I have no clue what is making it crashing at runtime and need to debug line per line :-/

user2684, 2020-05-09:
This should be fixed by #513 :-) (issue related to #463)

closed 2020-05-09 by user2684