mysensors/NodeManager #534: Motion sensor send only "Motion" event

KooLru, 2021-02-04:
Motion sensor send only "motion" event, and not send "no motion" after PIR sensor switch to LOW (I am use AM612).

I changed setInterruptMode(CHANGE); in construtor of SensorMotion and it works fine - send V_TRIPPED with value 0.

Motion sensor example from MySensor.Org use same behavior

Can you explain, what are you use only RISING mode, and how to determine "off" motion event ? Or may be I can pool request  for change RISING mode to CHANGE

user2684, 2021-02-09:
Hi, yeah, it was intentional indeed, remember this was discussed long time ago and we went for RISING as the best compromise. Even if I agree with you point of view I'd not change the default behaviour to avoid changing expectations to users already using the sensor and upgrading NodeManager. If by setting setInterruptMode(CHANGE) it works in the way you expect, I'd keep it as it is. Makes sense? Thanks

KooLru, 2021-02-24:
I add

**motion.setInterruptMode(CHANGE);**

in **before()** function

Thanks!

closed 2021-02-24 by KooLru