mysensors/NodeManager #544: Changes for saving childs value in EEPROM

j54n1n, 2021-07-29:
Hello there,

In search for issue #542 I noticed that saving a value of a sensor in the EEPROM of the microcontroller allows only a limited range for positive numbers up to 9999 and has similar issues with the decimal part of floating point numbers.

I have seen that the code uses following layout: `[TYPE][SIGN][INT_PART1][INT_PART2][DEC_PART1][DEC_PART2]`

I would like to propose that the packing and unpacking could be done in a different manner using bitshifting and pointer arithmetic with casts, since either a `uint16_t` or a 32-bit float gets saved in the EEPROM. This would mean that the SIGN byte could be omitted since it is saved in the numerb itself and the entire frame could be shortened to 5 bytes instead of 6. Also this would allow to save the entire range of integer and float numbers in the EEPROM.

The result would like be something like this:
For 16-bit ints `[TYPE][INT16_PART1][INT16_PART2][UNUSED][UNUSED]`
and for floats `[TYPE][FLOAT_PART1][FLOAT_PART2][FLOAT_PART3][FLOAT_PART3]`

I will try to publish a pull request with the mentioned changes.

Kind regards.

user2684, 2021-08-10:
Great idea, this was a known issue I never had the chance to look into :-) Just one question before merging it: what happens is a user upgrades NodeManager and was storing something in the EEPROM before? I guess it will get corrupted/wrong data right? Not a problem of course, I can always make this clear in the release notes when going GA with this version, but just to confirm my understanding is correct

j54n1n, 2021-08-16:
> Great idea, this was a known issue I never had the chance to look into :-) Just one question before merging it: what happens is a user upgrades NodeManager and was storing something in the EEPROM [...]

Yes the binary format changes, so for the first time after an upgrade there will be false data restored for both integer and float values and for each successive child ID sensor values stored in EEPROM, except if the sensor value type does not match then nothing will be restored. So mabye only the first stored sensor child value in EEPROM will restore false data.

But in my case I tested it with my weather station node and after an sensor data update it will be overwritten with the new value and layout.

For sure a note should be included for the next release.

user2684, 2021-08-20:
Thanks for the detailed explanation, makes sense to me! Definitely the advantage of using this approach overcomes the drawback of having a false read, once. I'll make it clear anyway of course in the release notes when this version will be officially released. Merging your PR into development right now. Thanks again for the effort!

closed 2021-08-20 by user2684