mysensors/NodeManager #551: SensorSHT31.h

NickLplus, 2022-05-21:
A small one:
SensorSHT31.h reports temperature in ° Fahrenheit even when I_CONFIG is 0 ('Metric').

user2684, 2022-06-06:
Hi, never seen this behaviour before, will mark it as a bug and try to reproduce it. In case you have logs to share would be great. Thanks

NickLplus, 2022-06-06:
Thank you for looking into it. The line
`temperature = nodeManager.celsiusToFahrenheit(temperature);`
seems to be executed always, without a check for I_CONFIG value.

user2684, 2022-06-11:
Actually the check is inside the function (https://github.com/mysensors/NodeManager/blob/master/nodemanager/Node.cpp#L138). The logs should tell you if I_CONFIG is updated correctly, let's have a look :-)